rxrbln/t2sde #2: Cross-compiled Perl lacks some modules

lenticularis39, 2021-07-22:
When cross-compiling Perl (tested on x86_64 -> hppa, might happen on other archs), the resulting package is much smaller than when native compiling:

```
bash-5.1# wc -l /var/adm/flists/perl
5684 /var/adm/flists/perl
bash-5.1# wc -l build/hypatia-21-svn-generic-hppa-cross-linux/var/adm/flists/perl 
1703 build/hypatia-21-svn-generic-hppa-cross-linux/var/adm/flists/perl
```

After examining the flists it's apparent that many modules are missing from the cross-compiler package (maybe a gcc cross compiling problem?), for example Scalar/Util.pm.

The full flists are attached to the issue, along with the cross compilation build log.

[cross.flist.log](https://github.com/rxrbln/t2sde/files/6865296/cross.flist.log)
[cross.log.gz](https://github.com/rxrbln/t2sde/files/6865297/cross.log.gz)
[native.flist.log](https://github.com/rxrbln/t2sde/files/6865298/native.flist.log)

rxrbln, 2021-07-23:
Does perl in t2/trunk currently even cross compile? ;-)
Decades ago this was by design as more did usually not cross compile, maybe now with the restructured 1 and 2 cross stages we could build a more full features perl in 2-perl?
If anyone using Perl wants to give modifying the perl.{desc,conf} a try for that and post the result it would be helpful.

lenticularis39, 2021-07-23:
It doesn't have the CROSS flag set, but unlike some time ago Perl itself cross compiles fine, it's only the bundled modules which fail. Having Perl cross compiled would help, because it's a common package which takes some time to build on older systems.

I'll probably have a look at it myself when I get to it, I reported the issue mostly to track it for myself :)

rxrbln, 2021-07-23:
IIRC I removed the CROSS flag recently as it did not cross build for me, and I did not wanted bootstraps to error out on perl. So from my experience Perl used to cross-build as patched or configured in t2 until recently. Though this native mini perl hack needed some re-diffing every other year anyways ;-)

lenticularis39, 2021-07-24:
After investigating the log it's clear that it's trying to execute the cross-compiled perl binary at some point:
```
./perl: 1: Syntax error: ")" unexpected
```
It's quite unbelievable that after so many years Perl still hasn't be able to make the build system cross-compile aware, requiring patches or hacks.