rxrbln/t2sde #11: cross compiiled glibc shoudl install locales, or postinstall

rxrbln, 2021-07-28:
Currently a cross compiled glibc is missing all the pre-compiled locales. We should either hack those to generate somehow, or run the most important ones (en_US.UTF-8?) postinstall.d

lenticularis39, 2021-07-28:
crosstool-ng simply [copies locales from native build](https://crosstool-ng.github.io/docs/caveats-features/), it's far from ideal though.

N0T4G, 2021-11-30:
Also on musl, by design, the locale always points out to the internal musl C.UTF-8 locale so we might add LANG="C.UTF-8" in the /etc/profile.d/locale as a default to the install menu when installing a musl target or something.

rxrbln, 2021-11-30:
> Also on musl, by design, the locale always points out to the internal musl C.UTF-8 locale so we might add LANG="C.UTF-8" in the /etc/profile.d/locale as a default to the install menu when installing a musl target or something.

cool, does glibc supports C.UTF-8?

N0T4G, 2021-12-28:
Support has been added to glibc and is stated to ship in 2.35 (aka next release).

Maybe we could try and import the 16 patches from this commit but meh:
https://sourceware.org/git/?p=glibc.git;a=commit;h=466f2be6c08070e9113ae2fdc7acd5d8828cba50

rxrbln, 2021-12-28:
yeah, I saw your message on discord and was thinking about it. But in the grand scheme of things I considered it too minor to break stuff now in the last minute when we just get it automatically next year. The 21.12 release will run localdef for the chosen language ;-)

N0T4G, 2024-02-29:
https://git.adelielinux.org/adelie/musl-locales/-/wikis/home