rxrbln/t2sde #17: Package GralVM

rxrbln, 2021-07-28:


lenticularis39, 2021-12-26:
GraalVM is actually just a marketing name for a group of related programs (some of them have a confusing name themselves) together with their binary distribution (including a package manager), it doesn't refer to a piece of software in particular. These are:

- GraalVM compiler - a JIT compiler for OpenJDK written in Java. Requires OpenJDK with JVMCI (i.e. at least 11)
- Native Image - an AOT compiler for Java
- Substrate VM - runtime used by executables generated by Native Image
- Truffle - framework for implementing other programming languages than Java to run under GraalVM compiler
- various language implementations for Truffle from Oracle (GraalJS, FastR, GraalPython, TruffleRuby, Sulong) - the last of these runs LLVM IR

Adding them thus requires getting OpenJDK working first (currently we only have old binary JDK packages up to Java 1.6).

lenticularis39, 2022-01-21:
Having OpenJDK packaged it's now possible to build GraalVM on T2 manually, although the build system stands in the T2 way of doing things by downloading along with Java dependencies a pre-built JDK from Oracle. Also the built jar files look different than those shipped with GraalVM binaries and I haven't figured out how to enable them within the T2 OpenJDK yet, so there's still work to do.

For anyone interested see oracle/graal on GitHub and the build system at graalvm/mx that is used to build the GraalVM packages.

lenticularis39, 2022-01-23:
Commited revision 55247.

There's still some room for improvement (fix Sulong/Espresso build, package Native Image and Truffle languages), but the main parts (JDK with GraalVM compiler and Truffle) work.