rxrbln/t2sde #27: package perf from linux kernel tools

rxrbln, 2021-07-29:
IMHO we should ship.a perf package, as found in the linux kernel sources (sigh) tools/perf. IMHO^2 this should already be a standalone package upstream proper, but it isn't. It might be a good idea to take a look what other gems are hiding in some linux kernel tools/ and other corners, ... ;-)

lenticularis39, 2021-08-02:
> IMHO we should ship.a perf package, as found in the linux kernel sources (sigh) tools/perf.

Fedora's 3000-line kernel.spec builds its perf together with the kernel. It should be possible to build it separately, though, it's just a Fedora habit to build more binary packages together from one source.

> It might be a good idea to take a look what other gems are hiding in some linux kernel tools/ and other corners, ... ;-)

I know about libbpf, which is synchronized using a shell script into a [separate repository](https://github.com/libbpf/libbpf), not sure about compatibility though, since it apparently follows bpf and bpf-next, not the mainline kernel.

rxrbln, 2022-01-05:
Committed revision 54900.

closed 2022-01-05 by rxrbln