rxrbln/t2sde #48: Provide some kind of docker or other container images

rxrbln, 2021-11-28:


IkerGalardi, 2021-11-30:
Is there any plan to get this soon? Want to start helping with some packages but getting a virtual machine up and running is a bit of a pain.

rxrbln, 2021-11-30:
I personally will likely not work on this before 2022, first we need to get the 21.12 release out and do some other stuff. Just running the release in a VM, e.g. in Qemu/KVM should be easy though. I do this daily, installs in just some minutes: http://t2sde.org/documentation/installintro.html , ... or extract the tarballs and chroot() it...

IkerGalardi, 2021-11-30:
Didn't realize the tarball thingy... Importing that tarball into a scratch docker image would basically create a T2 docker container right? Or am I missing something else?

mercuriete, 2021-11-30:
I tried to do a dockerfile but in the web threre are only a ISO.
I tried to extract stage2 folder from the ISO but I couln't.

Can you provide the link to those tar balls I only found source tarballs?

IkerGalardi, 2021-11-30:
As far as I know, the sources are all we need to compile and test new packages. Correct me if I'm wrong.

mercuriete, 2021-11-30:
for doing chroot to some kind of t2 distro we need a tar with binaries, not sources...
I think I can do a FROM scratch Dockerfile but first I need the compiled root filesystem.
As far as I know, that is not on the download page.

@rxrbln Can you provide the links to those tarballs that we can do chroot?

rxrbln, 2021-11-30:
the binary tarballs are in each ISO as used by the installer, simply loopback mount the matching architecture one and for loop extract those e.g.  /media/21.11-generic-x86-64-nocona-cross-linux/pkgs/ into your new container sysroot ;-)

N0T4G, 2021-12-01:
Took longer than I though but here it is: https://github.com/N0T4G/t2-docker-from-scratch

Just run `./build.sh` and you'll have a `t2sde` image created locally which you can then start with `docker run -i -t t2sde`.
Enjoy!

N.B.: I did not succeed accessing the internet from within the inside so please report if you have a solution to that ^^

mercuriete, 2021-12-02:
@N0T4G 
you can 
```
ADD url path
```
To download from internet.

N0T4G, 2021-12-02:
Ah no ok my bad! My host machine had a particular config which required me to do `docker run --network=host` but that shouldn't be the case for anyone else ^^

mercuriete, 2021-12-02:
@N0T4G hi again, I saw your code and I think you can use only docker for making the FROM scratch image.
I was having this idea but I couldn't finish:
```Dockerfile
FROM ubuntu:latest AS build

WORKDIR /usr/src

RUN apt-get update && apt-get install -y p7zip-full zstd

ADD https://dl.t2-project.org/binary/2021/t2-21.4-x86-64-minimal-desktop-gcc-glibc.iso image.iso

#ADD image.iso image.iso

RUN 7z x image.iso
RUN rm -rf image.iso
RUN mkdir stage2 && cp stage2.tar.zst stage2

WORKDIR /usr/src/stage2

RUN tar --use-compress-program=unzstd -xvf stage2.tar.zst
RUN rm stage2.tar.zst

FROM scratch

COPY --from=build /usr/src/stage2/ /

CMD ["/bin/bash"]
```

What do you think about using docker for extracting files from the iso file?

mercuriete, 2021-12-02:
finally I got my version working:

```Dockerfile
FROM ubuntu:latest AS build

WORKDIR /usr/src

RUN apt-get update && apt-get install -y p7zip-full zstd

ADD https://dl.t2-project.org/binary/2021/t2-21.4-x86-64-minimal-desktop-gcc-glibc.iso image.iso

RUN 7z x image.iso
RUN mkdir /usr/src/root

RUN find . -regex ".*/pkgs/.*\.tar\.zst" -exec sh -c 'tar --same-owner --use-compress-program=unzstd -xvf "$1" -C "/usr/src/root/"' sh {} ';' && chown -R root:root /usr/src/root/

FROM scratch

WORKDIR /

COPY --from=build /usr/src/root/ /

CMD ["/bin/bash"]

```

I hope somebody could enjoy this Dockerfile

N0T4G, 2021-12-28:
@mercuriete Right of course you can base your image on other distros but we wouldn't do that for an official image as we are an independent distribution. 😉

mercuriete, 2021-12-29:
@N0T4G 
That is not how it multistage dockerfiles work.

The image is not based on another distro.
You can search information about multistage dockerfiles here: https://docs.docker.com/develop/develop-images/multistage-build/

If you check carefully ubuntu is only used to decompress an ISO but on your code you need superuser privilegies to mount an ISO what is very unsecure. In this way everything is contained in a container and then squash in a single layer using FROM scratch.

N0T4G, 2021-12-29:
@mercuriete Fair enough 👌

Playing the devil's advocate here: what if the image you use as `build` is malicious (any image for that matter)?
Like maybe they have as special built `7z` that insert some compromised binary when they extract any distro's iso image?
I know that's a little far fetch but we never know ¯\_(ツ)_/¯

Isn't their a way to do what your dockerfile is doing but by using the hosts system tools (aka without using another image as the first stage?)

Anyway, thank you for your contribution. I'm sure that helped and will help many people! 🙂

N0T4G, 2024-02-23:
We would need to package a docker/container registry, to not rely on dockerhub.

Btw I recently commit `incus` (`lxc` fork) and `docker` #88 related packages to trunk. Haven't managed to run anything properly, some more debugging hours are required so if someone has some time try stuff out then patches welcome :)