rxrbln/t2sde #58: Compilation of the dialog tool failed

Kajanos, 2021-11-29:
Under Gentoo Linux;
`T440p ~/t2-trunk # ./scripts/Config 
Aquiring config write lock (may take until other configs finished)
Creating confdialog tool
cc misc/confdialog/*.c -Imisc/confdialog -lncurses -o src/confdialog.bin
misc/confdialog/confdialog.c: In function ‘j_inputbox’:
misc/confdialog/confdialog.c:230:25: warning: format not a string literal and no format arguments [-Wformat-security]
  230 |         fprintf(stderr, dialog_input_result);
      |                         ^~~~~~~~~~~~~~~~~~~
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: /tmp/ccd7Q6d2.o: undefined reference to symbol 'LINES'
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: /lib64/libtinfo.so.6: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
Compilation of the dialog tool failed, ncurses-devel missing?`

=================================================================
                        Package Settings
=================================================================

sys-libs/ncurses-6.3_p20211106::gentoo was built with the following:
USE="cxx gpm (split-usr) (tinfo) -ada -debug -doc -minimal -profile -static-libs -test -trace" ABI_X86="32 (64) (-x32)"
CFLAGS="-march=haswell -mtune=haswell -O3 -pipe -fgraphite-identity -floop-nest-optimize -fdevirtualize-at-ltrans -fipa-pta -fno-semantic-interposition -flto=8 -fuse-linker-plugin -falign-functions=32 -ldl -Wl,-O1"
CXXFLAGS="-march=haswell -mtune=haswell -O3 -pipe -fgraphite-identity -floop-nest-optimize -fdevirtualize-at-ltrans -fipa-pta -fno-semantic-interposition -flto=8 -fuse-linker-plugin -falign-functions=32 -ldl -Wl,-O1"
FEATURES="xattr config-protect-if-modified ccache multilib-strict merge-sync assume-digests strict binpkg-dostrip buildpkg userpriv parallel-fetch protect-owned sandbox binpkg-logs news fixlafiles qa-unresolved-soname-deps unknown-features-warn userfetch unmerge-orphans binpkg-docompress binpkg-multi-instance ebuild-locks usersandbox usersync sfperms collision-protect unmerge-logs network-sandbox pid-sandbox preserve-libs distlocks ipc-sandbox"
LDFLAGS="-Wl,-O1 -march=haswell -mtune=haswell -O3 -pipe -fgraphite-identity -floop-nest-optimize -fdevirtualize-at-ltrans -fipa-pta -fno-semantic-interposition -flto=8 -fuse-linker-plugin -falign-functions=32 -ldl"

rxrbln, 2021-11-30:
well, it usually builds on other distributions, and even macOS, too. Not really an idea what Gentoo is doing there, but if you find out let us know! In the meantime you could install T2 in a (e.g. Qemu) VM or extract the tarballs and chroot into such a sys root: https://github.com/rxrbln/t2sde/issues/48#issuecomment-982475712

y0rune, 2021-11-30:
Hi!
I changed in the `scripts/Config` due to when I compiled it I had the same error like @Kajanos.
```diff
yorune@Gentoo ~/git/dev/t2-trunk master $ svn diff scripts/Config
Index: scripts/Config
===================================================================
--- scripts/Config      (revision 54366)
+++ scripts/Config      (working copy)
@@ -99,7 +99,7 @@
        if [ -z "$oldconfig" -a ! -f src/confdialog.bin ]; then
                echo "Creating confdialog tool"
                command="cc misc/confdialog/*.c `
-                       `-Imisc/confdialog -lncurses -o src/confdialog.bin"
+                       `-Imisc/confdialog -lncurses -ltinfo -o src/confdialog.bin"
                echo "$command"
                if ! eval "$command.$$"; then
                        echo "Compilation of the dialog tool failed, ncurses-devel missing?"
 ```
 
 So I check the `pkg-config --libs ncurses` and it returns:
 ```
-lncurses -ltinfo
 ```

After that compilation I had another error:
```
yorune@Gentoo ~/git/dev/t2-trunk master $ scripts/Config -delete; scripts/Config
removed 'config/default/packages'
removed 'config/default/config'
removed 'config/default/lock.pid'
removed directory 'config/default'
Aquiring config write lock (may take until other configs finished)
Creating confdialog tool
cc misc/confdialog/*.c -Imisc/confdialog -lncurses -ltinfo -o src/confdialog.bin
misc/confdialog/confdialog.c: In function ‘j_inputbox’:
misc/confdialog/confdialog.c:230:25: warning: format not a string literal and no format arguments [-Wformat-security]
  230 |         fprintf(stderr, dialog_input_result);
      |                         ^~~~~~~~~~~~~~~~~~~
T2 21-svn configuration
Building src/luabash.so
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/lapi.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/lauxlib.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/lbaselib.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/lcode.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/ldblib.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/ldebug.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/ldo.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/ldump.o
lua/lauxlib.c: In function ‘luaL_loadfile’:
lua/lauxlib.c:578:4: warning: this ‘while’ clause does not guard... [-Wmisleading-indentation]
  578 |    while ((c = getc(lf.f)) != EOF && c != LUA_SIGNATURE[0]) ;
      |    ^~~~~
lua/lauxlib.c:579:5: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘while’
  579 |     lf.extraline = 0;
      |     ^~
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/lfunc.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/lgc.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/linit.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/liolib.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/llex.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/lmathlib.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/lmem.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/loadlib.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/lobject.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/lopcodes.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/loslib.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/lparser.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/lstate.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/lstring.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/lstrlib.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/ltable.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/ltablib.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/ltm.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/lundump.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/lvm.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/lzio.o
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/print.o
  LINK LIB  /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/lua.a
  C         /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/bash/luabash.o
  LINK DYN  /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/bash/luabash.so
/usr/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../x86_64-pc-linux-gnu/bin/ld: /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/lua/lua.a: in function `os_tmpname':
loslib.c:(.text+0x127e1): warning: the use of `tmpnam' is dangerous, better use `mkstemp'
INSTALL DYNLIB /home/yorune/git/dev/t2-trunk/src/Linux-x86_64/bash/luabash.so
scripts/Config: line 134: enable: dynamic loading not available
Failed to enable the Lua bash accelerator
```

lenticularis39, 2021-11-30:
You need to rebuild bash with `plugins` USE flag.

Kajanos, 2021-11-30:
Looks like my problem is solved, Thank you all.

Best regards.

closed 2021-11-30 by Kajanos