rxrbln/t2sde #59: Potentially rework build scripts/

rxrbln, 2021-11-30:
AFAIR the current T2 build scripts/ are structured like they are since the ROCK Linux 2.0 rewrite about 20 years ago, and where similar even before that. While I prefer not to change a running system and preserve compatibility, maybe two decades later it might be worth a rework to be more streamlined, modern and user friendly. Maybe refactoring this to one common frontend (instead of multiple scripts) with more modern argument parsing would be nice to have. Just to brainstorm: the top level script could be named t2 (or pkg ;-) with commands like:

./t2 build $pkg
./t2 download $pkg
./t2 build --target $target

This could potentially be combined with reworking the installed, see issue #57

N0T4G, 2021-11-30:
Indeed a more modern and user-friendly frontend is long overdue for a rock solid distribution!
I would lean towards a `t2` top-level more than `pkg`.

lenticularis39, 2021-12-07:
Or perhaps keep the scripts and make `pkg`/`t2` a frontend that in addition to emerging/updating packages also can manage the T2 source tree, e.g. mimicking apt-get:

`pkg update` -> `svn up`
`pkg upgrade` -> `scripts/Emerge-Pkg -system -missing=no`

The scripts would stay as the low level part, while `pkg` would be a user friendly high level tool.

rxrbln, 2021-12-07:
Certainly a huge fan of not breaking what works, not re-inventing the wheel and retaining historic compatibility. 

We will certainly should work carefully and slowly on this, gradually improving the status quo. Maybe improving Emerge first, and certainly #60 is more important to takle first, or at least with higher priority.

lenticularis39, 2022-01-29:
We could do something like this (to make it intuitive for users of apt, pkgin or FreeBSD's or OpenSolaris' pkg):

`pkg update` - runs `svn up` in `T2ROOT`, default `/usr/src/t2-trunk`
`pkg upgrade` - runs `scripts/Emerge-Pkg -system $@`
`pkg install` - runs `scripts/Emerge-Pkg $@` (install updates dependencies on other systems too, we don't have to default to  `-missing=only`)
`pkg remove` - runs  `mine -r $@`
`pkg search` - runs `scripts/Find-Pkg`

pkg being located in /usr/bin and shipped as a package/part of an existing package.

rxrbln, 2022-02-27:
maybe our in source frontend scripts should be named t2 and the installed end user package maintenance frontend could be a symlink to pkg.  Mostly to make it more "intuitive" for developers using t2 on other systems to build embedded Linux firmware not to confuse it with other "pkg" implementations, ...

paulwratt, 2022-02-28:
This would qualify for a new Version Release (as opposed to Point Release or abn Update Release)

I too lean towards the use or `t2`

rxrbln, 2024-06-25:
FreeBSD https://man.freebsd.org/cgi/man.cgi?query=pkg&sektion=8&apropos=0&manpath=FreeBSD+14.1-RELEASE+and+Ports uses pkg add | create |  install |  delete | upgrade | …

rxrbln, 2024-06-25:
./t2 build|install|remove pkg --cfg mips64 --target build/