rxrbln/t2sde #87: Attempting to compile on WSL throws error

Coderdreams, 2022-02-13:
I attempted to check the "cross compiling" option in the config but the error is thrown anyway:

== 19:57:48 =[0]=> Building kernel/linux-header [5.1 22-svn]
-> Compiler Cache enabled: 0.00% cache hits so far
-> Parallel build using 12 jobs
-> Reading package config: package/kernel/linux-header/linux-header.conf
-> Preparing build in src.linux-header.system.20220213.195747.21172.DESKTOP-DJ9A1QE
-> Building. Writing output to $root/var/adm/logs/0-linux-header.out
-> Compiler Cache Hits while pkg build: 61.18% (93 hits, 59 misses)
!> ====[lx_injectextraversion:169 (last $?=0)> cut -d '"' -f 2
!> ===[lx_injectextraversion:168 (last $?=0)> lx_kernelrelease=
!> ===[lx_injectextraversion:169 (last $?=0)> '[' -z '' -a 1 ']'
!> ===[lx_injectextraversion:170 (last $?=0)> make -j12 ARCH=x86 CROSS_COMPILE= HOSTCC=cc KCC=kcc .kernelrelease incl ..
!> You should not run uname during the build: -m
!> You should not run uname during the build: -m
!> Due to previous errors, no 0-linux-header.log file!
!> (Try enabling xtrace in the config to track an error inside the build system.)
== +00:00:15 =[0]=> Aborted building package linux-header
-> Unmounting loop mounts ...

Host computer: Windows 11 with WSL "Ubuntu 20.04.3 LTS"

Coderdreams, 2022-02-14:
Actually I think the problem is with Ubuntu 20.04, I tried compiling there without luck, either natively or with Freebsd's debootstrap, it never worked, Ubuntu 18.04 doesn't work either.

paulwratt, 2022-02-25:
Is it just that its never been tested on WSL?

Surely cross-compile works on non-WSL Ubuntu (isn't that the OS Rene uses for his build system)?

TechdudeGames, 2022-03-21:
I have tried before to compile T2 on native Ubuntu, and it doesn't always work the way you want it to. Weird errors would crop up when I tried building, ranging from minor errors such as not having libisl-dev installed, to ld spitting out R_X86_64_PC32 errors at random stages of the build. My advice is to build T2 underneath a VM using a release iso of T2. I use a VM of T2 using [T2 version 21.7 for x86_64](https://dl.t2sde.org/binary/2021/t2-21.7-x86-64-minimal-desktop-gcc-glibc.iso). I did have to use ./scripts/Emerge-Pkg from the T2 source to install python and perl once I installed the base system as the binaries were missing, but once I did that I've been able to build T2 without the hassles I had under Ubuntu.

Coderdreams, 2022-03-21:
Yes, building on Ubuntu natively doesn't work, it seems it lacks some tool because the bash scripts throw "file descriptor" errors, but it's impossible for me to find what tool is that.

rxrbln, 2022-03-21:
you / someone could test if this is fixed with  revision 56180.

Coderdreams, 2022-03-26:
Now it's compiling further after the update, it got stuck building llvm but I see that you are attempting to fix that now, so maybe this will finish successfully after that. It's great that you are doing a video debugging that because it will teach us to debug ourselves when we have problems.

rxrbln, 2024-02-18:
worked for me with T2 in WSL https://www.youtube.com/watch?v=3GV3NNOH0Mc

closed 2024-02-18 by rxrbln