rxrbln/t2sde #88: Package docker packages

rxrbln, 2022-02-17:
https://github.com/docker

janithcooray, 2023-10-28:
# Update

Build System
```
Linux t2 6.4.4-t2
```

Got the docker daemon ( moby / same thing they do at arch ) to build 
but encountered some issue running. related to [this issue](https://github.com/docker/cli/issues/2104)

Dependencies -
```
containerd
build-essentials ( general GNU )
go
-> runc module
git
libseccomp
systemd (instead of using cgroupfs)
python3+
-> jinja2 module
```

https://github.com/mk-fg/cgroup-tools

Issue - Unable to mount cgroupfs, after some lookups, seems it's not required if systemd is active. have some build errors when trying to build systemd. will keep looking 

The rest of the docker tools are Fully Open Source

Screenshots:

Docker D
[image]

Startup Error

[image]

N0T4G, 2024-02-28:
Added `CONFIG_CGROUP_BPF` which fixed the issue.
I think the issue can be closed now as things seems to be working (at least for basic, one container usage).

closed 2024-02-28 by rxrbln