rxrbln/t2sde #110: Package dart

rxrbln, 2022-12-11:
https://dart.dev

termdew, 2023-11-22:
I would like to work on this one, maybe it's a good issue for practicing packaging for T2 SDE.

rxrbln, 2023-11-22:
sure, though dart might be more complex and thus difficult than the libreoffice bounty you I believe comments on earlier. Would you still consider the libreoffice reps bound? Because somehow your comment disappeared when I wanted to comment on it. Did you delete it?

termdew, 2023-11-22:
Yeah, I deleted the comment, because I thought that one package would be better as a beginners task.If you say that Dart is more complex, then please assign me to the LibreOffice issue. Sorry for the confusion.

ian-moody, 2025-07-30:
I think this can be closed as dart was first packaged as of commit 79f29a6.