rxrbln/t2sde #120: Package gcj from gcc java fame

rxrbln, 2023-03-07:
The once Jave to native gcj gcc compiler was removed from gcc. It is at time nice to have in certain situations, so it would be nice to have a gjc t2 package of the last stable version of gcc in t2.

paulwratt, 2023-10-12:
you mean `gjc-9.3.0` (for `gcc-9.3.0` in [v9.0 source mirror](http://dl.t2sde.org/mirror/9.0/g/)?

rxrbln, 2023-10-12:
I think more like "GCJ was removed from the GCC trunk on September 30, 2016. Announcement of its removal was made with the release of the GCC 7.1, which does not contain it. GCJ remains part of GCC 6." -Wikipedia and downloaded straight from the gcc mirrors.

paulwratt, 2023-10-13:
Thanks for clarifying that. I have GCC6 compiled binaries from Jessie that still work on Buster today (without recompiling), so I for one would be happy having a complete set of v6 GCC compilers - would it be worth trying v7.0 (rhetorical question, for future readers)

terefang, 2024-03-18:
do i see that correctly, that it is for bootstrapping java.

iirc ... gcj-6 is required for ecj.jar plus jamvm to build 1.5 compatible classpath and iterate thru al the java versions

rxrbln, 2024-03-18:
not necessary for bootstrapping another Java, but still supporting the previous GCC based fallback of compiling Java to native executables. Decade ago gcj just built as part of gcc until they removed it some decade ago.

terefang, 2024-03-18:
i am currently on a journey to bootstrap java 
... so i can already say that gcc6 (musl-static) builds from gcc13
... have not had the time to try building gcj6 with it.