rxrbln/t2sde #127: google fonts are missing

terefang, 2023-07-19:
just seen your rant on open source fonts

maybe i can donate some code for you to adapt.

```bash
#!/usr/bin/env bash

XDIR=$(cd $(dirname $0) && pwd)

URL="https://fonts.google.com/download?family="

LIST=$(curl https://fonts.google.com/metadata/fonts 2>/dev/null | jq -r '.familyMetadataList[].family' |tr ' ' '+')

for x in ${LIST}; do

    NAME=$(echo $x | tr '+' '-')

        mkdir -p ${XDIR}/_tmp
        wget -O ${XDIR}/_tmp/${NAME}.zip ${URL}$x

        rm -rf ${XDIR}/${NAME}
        mkdir -p ${XDIR}/${NAME}
        cd ${XDIR}/${NAME} && unzip -o -j ${XDIR}/_tmp/${NAME}.zip

        rm -rf ${XDIR}/_tmp
done
```

N0T4G, 2024-02-23:
Hey @terefang awesome, thank you!
I will look into it as soon as possible

terefang, 2024-04-10:
i will make available bash based installers in a github repo under:

https://github.com/terefang/font-installer-scripts

maybe you want to package them instead ?

terefang, 2024-04-11:
the repo is ready

https://github.com/terefang/font-installer-scripts