rxrbln/t2sde #129: luajit switch to rolling releases

FractalU, 2023-09-08:
Luajit recently switched to rolling releases. It uses the unix timestamp from a commit as the last part of the version number. This is because timestamps give a unique, monotonically increasing number for each git commit. The version number looks like `2.1.1693350652`. The number `1693350652` is a unix timestamp. In addition, the `beta3` is gone. Luajit is officially not beta anymore. The git snapshots can continued to be used as before. Just the version number and the fact that Luajit is not beta anymore needs to be updated. The version number can be derived from the unix timestamp stored in the `.relver` file in the Luajit release tarball of the git snapshots. The command `luajit -v` can be used to check the version number.

Another thing I noticed is that in the Luajit package page of T2 the download link of the release tarball of git snapshots is broken up into two parts, even though in the source file `luajit.desc` that link is whole. Seems a bug in the auto generation.

Youtube apparently decided to delete my comment. That's why I'm writing here.

FractalU, 2023-09-10:
> Another thing I noticed is that in the Luajit package page of T2 the download link of the release tarball of git snapshots is broken up into two parts, even though in the source file luajit.desc that link is whole. Seems a bug in the auto generation.

Actually, it turns out to be not a bug but deliberate design. The download link of all other T2 packages are also broken up into two parts. One for the file and one for its parent directory. In very rare cases, only the file is accessible but not its parent directory. It happened to be the case with the download links to git snapshots. I don't think its a bother because there are so many more download links that lead to `404 not found`. From a UI perspective, splitting up links that way is not well hinted. The only way to see this is to recognize that there is a small gap between the link of the parent directory with a trailing slash and the filename.

FractalU, 2024-09-28:
@rxrbln Have you already tried to implement an auto updater for Luajit? I checked the Luajit package on T2. The version is still 2.1-9b5e837. See no progress there.

rxrbln, 2024-09-28:
it makes no sense for operating systems to bi-daily update a package, let alone something as rarely used as luajit to the latest authors personal git tree version. Instead major, tested release versions are shipped. In case of software maintained with so little effort as to even put out proper point releases such as luajit we would manually cherry pick a known good release every know and then when needed or major bugs or features were implemented. The next time we update the luajit pkg we will close this bug. Thank you very much.