rxrbln/t2sde #130: scripts/Config: lua bash error: unknown/unsupported bash version (Error on Debian 10)

docelic, 2023-09-26:
Heya,

If you could shed some light on this it'd be appreciated. This is on Debian 10:

```sh
/tmp/t2-8.0$ ls -al /bin/sh
lrwxrwxrwx 1 root root 4 Sep 22 18:24 /bin/sh -> bash
# (Manually changed from dash to bash)

/tmp/t2-8.0$ ls -al /bin/bash
-rwxr-xr-x 1 root root 1168776 Apr 18  2019 /bin/bash

/tmp/t2-8.0$ /bin/bash --version
GNU bash, version 5.0.3(1)-release (x86_64-pc-linux-gnu)
```

When trying to open Config:

```sh
/tmp/t2-8.0$ ./scripts/Config 
Aquiring config write lock (may take until other configs finished) ...
Creating confdialog tool.
gcc misc/confdialog/*.c -Imisc/confdialog -lncurses -o src/confdialog.bin
T2 8.0-stable configuration ...
Building src/luabash.so.
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/lapi.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/lauxlib.o
lua/lauxlib.c: In function ‘luaL_loadfile’:
lua/lauxlib.c:578:4: warning: this ‘while’ clause does not guard... [-Wmisleading-indentation]
    while ((c = getc(lf.f)) != EOF && c != LUA_SIGNATURE[0]) ;
    ^~~~~
lua/lauxlib.c:579:5: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘while’
     lf.extraline = 0;
     ^~
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/lbaselib.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/lcode.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/ldblib.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/ldebug.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/ldo.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/ldump.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/lfunc.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/lgc.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/linit.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/liolib.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/llex.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/lmathlib.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/lmem.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/loadlib.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/lobject.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/lopcodes.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/loslib.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/lparser.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/lstate.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/lstring.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/lstrlib.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/ltable.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/ltablib.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/ltm.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/lundump.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/lvm.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/lzio.o
  C         /tmp/t2-8.0/src/Linux-x86_64/lua/print.o
  LINK LIB  /tmp/t2-8.0/src/Linux-x86_64/lua/lua.a
  C         /tmp/t2-8.0/src/Linux-x86_64/bash/luabash.o
  LINK DYN  /tmp/t2-8.0/src/Linux-x86_64/bash/luabash.so
/usr/bin/ld: /tmp/t2-8.0/src/Linux-x86_64/lua/lua.a: in function `os_tmpname':
loslib.c:(.text+0xf340): warning: the use of `tmpnam' is dangerous, better use `mkstemp'
INSTALL DYNLIB /tmp/t2-8.0/src/Linux-x86_64/bash/luabash.so
lua bash error: unknown/unsupported bash version
```

Thanks!

paulwratt, 2023-10-12:
that v5.0.3(1) is the same version as I have on RPiOS 10 (Buster), but if you look in the [T2SDE source mirror](http://dl.t2sde.org/mirror/trunk/b/), `bash` is version 5.2, so that maybe why you have a conflict

docelic, 2023-10-13:
Good point, I should use 9.0 in any case. Thanks.

closed 2023-10-13 by docelic