rxrbln/t2sde #132: Error relocating /usr/lib/libfl.so.2: yylex: symbol not found

svmhdvn, 2023-10-14:
Building the following packages (probably a few other packages as well, I'll edit this description as I find them) fails on a fresh install of `t2-23.10-x86-64-minimal-firefox-gcc-musl-nocona.iso`:
* git
* python

Same message shows up:
`Error relocating /usr/lib/libfl.so.2: yylex: symbol not found`

scsijon, 2023-10-14:
On 14/10/23 14:11, Siva Mahadevan wrote:
> Building the following packages (probably a few other packages as well, I'll edit this description as I find them) fails on a fresh install of `t2-23.10-x86-64-minimal-firefox-gcc-musl-nocona.iso`: [...]
There are a few of these if you do a search on the error. They all seem 
to point back to the one problem that seems to be clearest in this > 
https://copyprogramming.com/howto/undefined-reference-to-yylex.

paulwratt, 2023-11-15:
you cant build anything on that image, because of that error

before you get to that error however, you will be missing `getconf` - musl-utils contains a couple of other missing items from that image as well (`getent` & `iconv`)

paulwratt, 2023-11-15:
but you cant build that either, because `yylex: symbol not found` ...

seems like the only _real_ way to fix this is to patch `libfl` at the time the ISO is built from source, to use `flex-2.6.x`

paulwratt, 2023-11-15:
it seems surprising that this was not found before now (in any `musl` build ?)