rxrbln/t2sde #134: your Distro Name T2 is conflicting with T2 Linux

juanpc2018, 2023-10-16:
[T2 Linux](https://t2linux.org/) is an effort to make [Apple T2 chip](https://en.wikipedia.org/wiki/Apple_T2) Intel Macs work with [Linux](https://github.com/t2linux).

rxrbln, 2023-10-16:
That's not our fault, we were first around 2003. Apple, Sun and other IT companies could have be more conscious and named their temporary (and internal product name not widely advertised) and in the meantime obsolete products differently. T2 Linus folks could also have been wiser and chosen a non colliding name like the Asahi Linux folks. Alas, Apple's T2 chips always sucks and are now thankfully obsolete. ;-)

juanpc2018, 2023-10-16:
But if your Distro is based on [Rocky Linux](https://rockylinux.org/), 
should be called R2?

rxrbln, 2023-10-16:
Thank you for your suggestions. That's all nice and fun. However, we'll not going to change the name without very good reasons jeopardizing the small name and following built ourself up. If we ever change our name it would need to be some AAA, big business compatible name. We are not based on Rocky Linux. We are the continuous of ROCK Linux, something entirely different from Rocky Linux (another very bad name choice TBH). A unique from scratch Linux distribution founded as far back as 1996-ish, not having any root or shared code base with any other Linux distribution. E.g. not a fork of a fork of yet another fork of Debian, RedHat or Slackware.