rxrbln/t2sde #135: rocknet: killall fails

paulwratt, 2023-10-28:
there is actually a bug in `rocknet` where if you shut down _any_ interface (or both via `/etc/init.d/network`) it will try to:
```
killall -TERM dhclient
```

Of coarse that should shut down _only_ the process associated with the associated _interface_.

It also appears that the version of `killall` includes with the 23-10 build _does not support_ `-TERM`, however `-s SIGTERM` works.

```
# killall --version
killall (PSmisc) 23.6
```

paulwratt, 2023-10-28:
I have a working fix using `kill` instead, which will `SIGTERM` any process with a given interface, but I introduced another function to allow for only _one_ `dhclient` instance per interface. ie. it closes any before before starting another.

This is OK tho, because I found that after a couple of hours with no endpoint, the `dhclient` process would stall.

paulwratt, 2023-10-31:
I had a look at the change you made, using a PID. The only issue is if someone is doing things from the command line, and not through `rocknet`. The changes I made cater for that too.

----

I did manage to get `sftp` working, so I will post something tonight (when I can plug in again)

paulwratt, 2023-11-01:
```
Index: rocknet_dhcp.sh
===================================================================
--- rocknet_dhcp.sh	(revision 65522)
+++ rocknet_dhcp.sh	(working copy)
@@ -2,7 +2,7 @@
 # This copyright note is auto-generated by ./scripts/Create-CopyPatch.
 # 
 # T2 SDE: package/.../dhcp/rocknet_dhcp.sh
-# Copyright (C) 2004 - 2019 The T2 SDE Project
+# Copyright (C) 2004 - 2023 The T2 SDE Project
 # 
 # More information can be found in the files COPYING and README.
 # 
@@ -12,8 +12,20 @@
 # GNU General Public License can be found in the file COPYING.
 # --- T2-COPYRIGHT-NOTE-END ---
 
+dhclient_start () {
+	local if="$1"; shift
+	dhclient_stop $if
+	ip link set $if up || ifconfig $if up
+	sleep 1 ; /sbin/dhclient -nw "$@" $if
+}
+
+dhclient_stop () {
+	local if="$1"; shift
+	ps -C dhclient o pid,args --no-headers | grep $if | cut -d \  -f 2 | xargs -I {} kill -s SIGTERM {}
+}
+
 public_dhcp() {
-	addcode up   5 1 "ip link set $if up"
-	addcode up   5 5 "dhclient -q -pf /var/run/dhclient-$if.pid $@ $if"
-	addcode down 5 5 "/sbin/dhclient -pf /var/run/dhclient-$if.pid -x"
+	addcode up   5 5 "dhclient_start $if $*"
+	addcode down 5 5 "dhclient_stop $if"
+	addcode down 5 6 "sleep 1; ip addr flush $if && ip link set $if down || ifconfig $if down"
 }
```

---

I just did a `svn up` too

rxrbln, 2023-11-01:
thank you for the example code. As mentioned above I'd rather keep base level system scripts KISS, Keep It Stupid Simple. Your code also only works with a full featured version of ps (guess not busy box or toxbox, or other alternatives) and on a first glance overmatches $if if you have more than 9, like enterprise servers often do, ... E.g. eth1 vs eth10 and so on. There are good reasons why Unix services historically used pid file to keep track of daemons started.

paulwratt, 2023-11-01:
OK, not that I have not upgraded any software _post install_. so that full PS is part of the _base_ . Do you have any BusyBox / ToyBox T2 ISO's I can test this against.

I could test a combined one (I might be able to _hack_ it to test on my Alpine server, where they already use PIDs for _everything_ )

paulwratt, 2023-11-01:
RE: that unix problem, I changed the above patch from `grep $if"` to `grep $if"\$"`

paulwratt, 2023-11-01:
RE: full PS. `ps -C` comes from a `whichps` scxript I use on a daily basis, and if I pass it _TWO_ arguments, it actually does not use `ps -C`:

```
#!/bin/sh
ATR_BOLD="$(tput bold)"
ATR_NONE="$(tput sgr0)"
COLS="$(tput cols)"
LIMIT="$(expr $COLS + $COLS)"
if [ "$2" = "" ];then
  echo "type: ${ATR_BOLD}ps -C "$1" o pid,args --no-headers${ATR_NONE}"
  ps -C "$1" o pid,args --no-headers
else
  echo "type: ${ATR_BOLD}ps aux | grep "$1" | grep "$2"${ATR_NONE}"
  ps aux | cut --bytes=-$LIMIT | grep -v "$0 $1 $2" | grep "$1" | grep "$2" | grep -v "grep $2"
fi
```

paulwratt, 2023-11-01:
so based on the above script, the new PID grabber should looke something like:
```
+	COLUMNS=255 ps aux | grep -e dhclient -e $if"$" | cut --byte-limit=-5 | xargs -I {} kill -s SIGTERM {}
```
(presuming here that BusyBox, ToyBox and BSD can also `grep -e`)

paulwratt, 2023-11-01:
so to throw your PID fix back into the mix, the new diff should look something like:

```
Index: rocknet_dhcp.sh
===================================================================
--- rocknet_dhcp.sh	(revision 65522)
+++ rocknet_dhcp.sh	(working copy)
@@ -2,7 +2,7 @@
 # This copyright note is auto-generated by ./scripts/Create-CopyPatch.
 # 
 # T2 SDE: package/.../dhcp/rocknet_dhcp.sh
-# Copyright (C) 2004 - 2019 The T2 SDE Project
+# Copyright (C) 2004 - 2023 The T2 SDE Project
 # 
 # More information can be found in the files COPYING and README.
 # 
@@ -12,8 +12,20 @@
 # GNU General Public License can be found in the file COPYING.
 # --- T2-COPYRIGHT-NOTE-END ---
 
+dhclient_start () {
+	local if="$1"; shift
+	dhclient_stop $if
+	ip link set $if up || ifconfig $if up
+	sleep 1 ; /sbin/dhclient -pf /var/run/dhclient-$if.pid "$@" $if
+}
+
+dhclient_stop () {
+	local if="$1"; shift
+	[ -f /var/run/dhclient-$if.pid ] && /sbin/dhclient -pf /var/run/dhclient-$if.pid -x
+	COLUMNS=255 ps aux | grep -e dhclient -e $if"$" | cut --byte-limit=-5 | xargs -I {} kill -s SIGTERM {}+}
+
 public_dhcp() {
-	addcode up   5 1 "ip link set $if up"
-	addcode up   5 5 "dhclient -q -pf /var/run/dhclient-$if.pid $@ $if"
-	addcode down 5 5 "/sbin/dhclient -pf /var/run/dhclient-$if.pid -x"
+	addcode up   5 5 "dhclient_start $if $*"
+	addcode down 5 5 "dhclient_stop $if"
+	addcode down 5 6 "sleep 1; ip addr flush $if && ip link set $if down || ifconfig $if down"
 }```

closed 2023-11-13 by rxrbln