rxrbln/t2sde #137: sshd: /etc/init.d/sshd "borks" on restart if config is not legal

paulwratt, 2023-10-31:
I mad a change to test for a legitimate `sshd_config` before `sshd` _start_ , but also _don't_ kill process on _restart_ if the config is not legal.

----

I post the change tonight (in about 6-8 hours)

rxrbln, 2023-11-01:
Well, most people only run a dhcp client per machine, which is probably why this was never an issue for 20 years since the ROCK Linux days. As usual I think we should keep it stupid simple, and my fix fixes the SIGTERM, as well as elegantly using the dhclient provided means even a dhclient per interface. Both nice improvements over the 20yo state. I don't think we should try to optimize for mixing rocknet and manual fiddling. This will usually anyway not match nor work, so we should not clutter the code necessarily. IMHO if one manually fiddles around, like I myself for testing, one can also just manually clean up their stuff.

Actually one thing I would re-write in rock net is saving the ifup state, and using that for ifdow. Currently the configurationstate is always re-evaluated so that editing the configuration between ifup and if down results in if down using newly edited and thus potentially not matching commands. No a daily issue, but certainly another nice improvement to be done eventually just for such cases.

paulwratt, 2023-11-01:
You probably saw me ask about this on Discord, but here is the `svn di`. 

While trying to get SSHd to to work for me, I missed a _value_ from one of the _options_ and the init script " _spat the dummie_ " (_borked_) and then killed the SSHd process, which  I am figuring ina production server is **_NOT_** something you want.

In this version I chose to leave  `-T > /dev/null` in (instead of `-t` ), just incase someone actually wants to see parsed config out from the init.

```
Index: sshd.init
===================================================================
--- sshd.init	(revision 65522)
+++ sshd.init	(working copy)
@@ -17,6 +17,13 @@
 main_begin
 
     block_begin(start, `Starting sshd')
+	/usr/sbin/sshd -T > /dev/null || error=$?	
+	if [ ! $error -eq 0 ]; then
+		echo "Can't start sshd: Error in sshd_config !!"\
+			"Please edit sshd_config and verify"
+		status
+		exit
+	fi  
 	if ! test -e /etc/ssh/ssh_host_key -o \
 		  -e /etc/ssh/ssh_host_rsa_key -o \
 		  -e /etc/ssh/ssh_host_ecdsa_key -o \
@@ -39,6 +46,13 @@
     block_end
 
     block_begin(restart, `Restarting sshd')
+	/usr/sbin/sshd -T > /dev/null || error=$?	
+	if [ ! $error -eq 0 ]; then
+		echo "Can't start sshd: Error in sshd_config !!"\
+			"Please edit sshd_config and verify"
+		status
+		exit
+	fi  
 	check(`if [ -s /var/run/sshd.pid ]; then
 		kill -HUP $(cat /var/run/sshd.pid)
 	else
```

paulwratt, 2023-11-01:
I asked about this on Discord, because I think there might be a way to _thin it down_ or fit in into a `check` function, except that it needs to `exit` if it fails, _after doing a `status` (to show the **FAIL** )