rxrbln/t2sde #147: gcc 13.2.0 compilation failing

run-stop, 2024-01-11:
Hello Everybody,

I'm building the Live ISO image using the generic instructions (quick build). I'm using a recent VM running the recent T2 installation. During the building process, I'm facing the following error while building the GCC:

`  Stack-Smashing Protector disabled for this package                                                                                                                                
  Reading build/.../cross/lib/pkg_gcc_pre.conf                                                                                                                                      
  Reading build/.../cross/lib/pkg_gcc_post.conf                                                                                                                                     
  Building in src.gcc.system.20240111.112636.2165.dhcp, with 16 threads                                                                                                
  Writing output to $root/var/adm/logs/0-gcc.out                                                                                                                                    
! mkdir -p -- rust/.deps                                                                                                                                                            
! mkdir -p -- c-family/.deps                                                                                                                                                        
! mkdir -p -- common/.deps                                                                                                                                                          
! mkdir -p -- analyzer/.deps                                                                                                                                                        
! mkdir -p -- rtl-ssa/.deps                                                                                                                                                         
! config.status: executing default commands                                                                                                                                         
! Due to previous errors, no 0-gcc.log file!                                                                                                                                        
! (Try enabling xtrace in the config to track an error inside the build system.)                                                                                                    
  +00:00:23 Aborted building develop/gcc                                                                                                                                            
  Unmounting loop mounts ...
`

To my understanding, this is the dependency error, but based on the report (and the full error log), I don't see a single "error" or "failure" message. Typically, the error log will display the information about the missing header or library, but in this case, it looks clean (except for the error above).

Do you have any recommendations on how to fix this?

run-stop, 2024-01-12:
To fix the issue, create three files in the root of the source tree (see below), and the run the `Build-Target`:

```
package/develop/gcc/libiberty-fibheap.patch
```
```
--- gcc-13.2.0/biberty/fibheap.c-orig    2024-01-12 11:09:08.359000000 +0100
+++ gcc-13.2.0/libiberty/fibheap.c 2024-01-12 11:09:35.987000000 +0100
@@ -19,6 +19,8 @@
 the Free Software Foundation, 51 Franklin Street - Fifth Floor,
 Boston, MA 02110-1301, USA.  */
 
+#include "libiberty.h"
+#include "fibheap.h"
 #ifdef HAVE_CONFIG_H
 #include "config.h"
 #endif
@@ -31,8 +33,6 @@
 #ifdef HAVE_STRING_H
 #include <string.h>
 #endif
-#include "libiberty.h"
-#include "fibheap.h"
 
 
 #define FIBHEAPKEY_MIN LONG_MIN
```
```
package/develop/gcc/libiberty-filedescriptor.patch
```
```
--- gcc-13.2.0/libiberty/filedescriptor.c-orig   2024-01-11 15:14:09.164000000 +0100
+++ gcc-13.2.0/libiberty/filedescriptor.c        2024-01-11 15:15:06.262000000 +0100
@@ -23,6 +23,11 @@
 #include "ansidecl.h"
 #include "libiberty.h"
 
+#define HAVE_UNISTD_H
+#ifdef HAVE_UNISTD_H
+#include <unistd.h>
+#endif
+
 #ifdef HAVE_FCNTL_H
 #include <fcntl.h>
 #endif
```
```
package/develop/gcc/libiberty-libiberty-h.patch
```
```
--- gcc-13.2.0/include/libiberty.h.orig 2023-07-27 10:13:07.000000000 +0200
+++ gcc-13.2.0/include/libiberty.h      2024-01-12 11:32:28.906000000 +0100
@@ -41,6 +41,11 @@
 
 #include "ansidecl.h"
 
+#include <fcntl.h>
+#ifndef HAVE_LIMITS_H
+#define HAVE_LIMITS_H 1
+#endif
+
 /* Get a definition for size_t.  */
 #include <stddef.h>
 /* Get a definition for va_list.  */
```

rxrbln, 2024-01-12:
Thanks for the update. Interesting that such build error did not happen to me. Please note that the "livedcd" output was not used for a decade and does not just work due to various upstream and t2 changes since then. Based on your report I yesterday worked 2+h live on YouTube and Twitch to make it work again and will commit the livecd changes later today. I will probably also rename it from "livecd" to just "live" so you will have to update your config then.

run-stop, 2024-01-12:
Thank you @rxrbln! Can you please share the link to the Twitch session with me? 
What I'm probably missing is the correct initial environment to build the "live" environment.

It would be nice to have a guide on building the initial environment, so that process is reproducible.

rxrbln, 2024-01-12:
it's probably mostly this stream: https://www.twitch.tv/videos/2029406711 maybe the beginning was in: https://www.youtube.com/watch?v=zjCOjxBm-d8

with initial environment you probably mostly mean default login / password and starting some applications, like X or Wayland? That would be in the target rootfs/ overlay to copy / patch some initial files.

run-stop, 2024-01-12:
Actually, by initial environment I refer to the OS that you use to build the livecd. I tried to build it on Ubuntu 22.04 - it did not work. Then I used the most recent T2 image, installed in the VM, and used that. Somehow this gives a different set of issues (as reported).

Speaking of problems, the original URL for the Dev86src-0.16.21.tar.gz package is invalid. This one works: [Dev86src](https://github.com/anchorz/dev86/blob/master/Dev86src-0.16.21.tar.gz).

rxrbln, 2024-01-13:
vastly modernized and fixed renamed live(cd) target output:

r66456 | rene | 2024-01-13 11:19:31 +0100 (Sat, 13 Jan 2024) | 2 lines
        * renamed target/share/livecd -> live, remove @ from boot label
------------------------------------------------------------------------
r66455 | rene | 2024-01-13 11:15:21 +0100 (Sat, 13 Jan 2024) | 2 lines
        * removed debugging exec /bin/sh livecd leftover
------------------------------------------------------------------------
r66454 | rene | 2024-01-13 11:14:07 +0100 (Sat, 13 Jan 2024) | 2 lines
        * fixed initrd target output initrd compressor, likewise
------------------------------------------------------------------------
r66453 | rene | 2024-01-13 11:09:01 +0100 (Sat, 13 Jan 2024) | 5 lines
        * modernize and fixed shared target livecd output
        * concat right init
        * use modern, upstream overlayfs instead of unionfs / aufs
        * one-shot overlay w/o symlink mess, removes init2
------------------------------------------------------------------------
r66452 | rene | 2024-01-13 11:05:55 +0100 (Sat, 13 Jan 2024) | 2 lines
        * dont hard code and force size optimization in target livecd output
------------------------------------------------------------------------
r66451 | rene | 2024-01-13 11:04:47 +0100 (Sat, 13 Jan 2024) | 6 lines
        * fixed livecd target output:
        * for new initrd compressor
        * for /bin/sh not bash
        * guard for non-existing rootfs overlay
        * fix whitespacing

rxrbln, 2024-01-13:
> Actually, by initial environment I refer to the OS that you use to build the livecd. I tried to build it on Ubuntu 22.04 - it did not work. Then I used the most recent T2 image, installed in the VM, [...]

In general building T2 on other Linux OS should work. However, T2 is best to build t2 on t2 as distributions often do their own quirks or just missing dependencies. But patches welcome if you find concrete issues or bugs to improve building t2 on other Linux as we have initial support enablement to use t2 to build foreign homebrew on BSDs, macOS and Haiku, ...

closed 2024-02-29 by rxrbln