rxrbln/t2sde #163: Feature Suggestions

juanpc2018, 2024-04-30:
Problem #1.
OSX has a time saving feature, 
select all downloaded files, create a folder, 
File Manager moves automatically all files inside.

pearOS Monterrey has a modified KDE for 20.04 LTS
but pearOS decided to quit Ubuntu/Debian, and switch to Arch in later NiceC0re version.
Neon KDE was Not updated.

Problem #2.
Temperature SMART sensing...

New cheap M.2 drives have low energy efficiency = heat 
300MB/s per Watt.
example 2TB PNY CS3140
Good drives have 500MB/s Watt.

OS FileManager should detect SMART temperature,
and Pause automatically all transfers to avoid NAND overheat damage.

i dont have a M.2 to install T2 24.5 because got damaged.

juanpc2018, 2024-08-19:
it turns out that Nautilus File manager has those features, similar to OSX, 
pearOS Monterrey File Manager is based on Nautilus File Manager...

for some things: Dolphin is Easy/Bettter,
for other things: Nautilus is Better.
2 in 1 would be awesome.

adam-devel, 2024-12-21:
T2SDE is a Linux dstribution, not a desktop environment with a suite of tools. this is the wrong place to request these features. Post your requests to upstream devs (The Linux kernel, Gnome, KDE) or implement features yourself or pay someone to do it.

the first request is available in gnome's file manager, also known as "nautilus"

foxdrodd, 2026-01-04:
Thank you for taking the time to share your suggestions! Unfortunately I am not sure if this is the right repository for these, as I can't really connect them to T2SDE as a Linux Distro. But feel free to try and test T2SDE when you want.