rxrbln/t2sde #165: The minimal 24.5 installation doesn't work without lvm2 package

co-stig, 2024-05-02:
When installing from `t2-24.5-x86-64-base-wayland-glibc-gcc-nocona.iso`, I selected gasgui > "Only a minimalistic system" and didn't (de)select any packages manually, i.e. installed the default minimal set of packages. On the next step, stone couldn't write GRUB2 to the boot record, complaining that it couldn't load `libdevicemapper.so`. I then installed the `lvm2` package, and this made GRUB2 work.

Overall, it looks like `lvm2` package needs to be in the "minimal" set for the installation to finish correctly.

rxrbln, 2024-05-08:
Thank you for your report. I was just thinking some days ago I never use the minimal selection in years. The fix is easy to mark a handful of new required dependencies as base/ [C] category. I'll commit that in some days for the next release. I might take some extra time to make minimal not install Xorg anymore.

rxrbln, 2024-05-21:
thanks again for the report. This was fixed w/ r69047 and I even added a new minimal w/o xorg option while at it: r69028

closed 2024-05-21 by rxrbln