rxrbln/t2sde #167: Trying to get started: Dependency management

maximilian-tech, 2024-05-06:
I am very interested in this distribution, but I have troubles getting started and understanding how do use it.

I'd like to build `generic-base-wayland` with `znver2`  optimization.

I start from the base image: [t2-24.5-x86-64-base-wayland-glibc-gcc-nocona.iso](https://dl.t2sde.org/binary/2024/t2-24.5-x86-64-base-wayland-glibc-gcc-nocona.iso)

Then I execute 
```
svn co https://svn.exactcode.de/t2/tags/24.5 t2-24.5
cd t2-24.5
./scripts/Config
<Modify> set 'Base Wayland Desktop'
<Modify> set 'nocona' to 'zenver2'
< Exit >
./scripts/Build-Target
...
Building 0-base/glibc
...
! configure: error:
! *** python
...
...

```
So `python` is missing. But `python` is a dependency of `glibc`. Why isn't `python` build before?

Or do I use __T2__ in a wrong way? Am I missing crucial steps?

Thanks!

rxrbln, 2024-05-07:
Hi, yes, indeed python had a convoluted history and I guess there should be selftest diagnostics to check for that. Long story short: python does not 100%ly working cross compiled. We tried to patch it multiple times over the years but it was an ever breaking mess. As all our modern ISOs must be 100% cross compiled we can not ship python by default (patches welcome). The fresh install 101 points out installing python: https://t2sde.org/kb/8/ which should fix this case.

We tried bootstrapping python for probably quite a decade as part of the t2 target, but due to modern python dependency hell that lead to even more problems over time. For a user Emerge will install it into the system, but a target dev int needs to be installed in the system. Ideas and patches to improve all of this welcome!

maximilian-tech, 2024-05-07:
Thanks for the quick answer and pointing me to[https://t2sde.org/kb/8/]( https://t2sde.org/kb/8/)!

After executing
```
$ ./scripts/Emerge-Pkg -deps=none -f perl perl-xml-parser \
python pip setuptools meson ninja libtool libxml 
```
like described and additionally executing
```
./scripts/Emerge-Pkg ed
```
for `bc`, I am currently in build phase 1.
However, `Building 1-base/ncurses (6.4)` fails for some reason (ERROR-LOG has not Error or alike, the libs are also installed to the `build/<default-...-linux>/lib64`.

Can you point me to a configuration, does build? Is it better to install the Desktop/Wayland once the system is running natively?

---

Is the following process the correct one to use T2?:
`~[A]~`
1. Start from [t2-24.5-x86-64-base-wayland-glibc-gcc-nocona.iso](https://dl.t2sde.org/binary/2024/t2-24.5-x86-64-base-wayland-glibc-gcc-nocona.iso)
2. Execute `$ svn co https://svn.exactcode.de/t2/tags/24.5 t2-24.5 && cd t2-24.5` (or `trunk` instead of `24.5`)
3. Install needed packages (not included due to cross-compile shenanigans): $ ./scripts/Emerge-Pkg -deps=none -f perl perl xml-parser python pip setuptools meson ninja libtool libxml ed`
4. Configure the system via `$ ./scripts/Config` ( e.g. `generic-base-wayland` with `znver2` optimization.)
5. Execute `$ ./scripts/Build-Target`   <<--- Currently stalled here!
6. Execute `$ ./scripts/Create-ISO`
7. Install __T2__ via the created image.

`~[B]~`
One this `optimized` __T2__ is up an running, on should again 
1. Execute `$ svn co https://svn.exactcode.de/t2/tags/24.5 t2-24.5 && cd t2-24.5` (or `trunk` instead of `24.5`)
2. Install needed packages (not included due to cross-compile shenanigans): $ ./scripts/Emerge-Pkg -deps=none -f perl perl xml-parser python pip setuptools meson ninja libtool libxml ed`
3. Install any other packages via `./scripts/Emerge-Pkg` (e.g. `git`). Doing this is here the same as specifying them in the `~[A]~`4. via `X git` in the modified package selection(__?__).

__My question__: Say, I'd like a Desktop like Ubuntu (GNOME based), with `qt5base` package installed, how do I do this?
1. Continue from `~[B]~`3. and do ` ./scripts/Emerge-Pkg gnome-desktop qt5base`? 

Thanks again for the work you put into T2!

rxrbln, 2024-05-08:
do you want to build your own iso for distribution or just use it on your desktop? If you only want to use it, you only need scripts/Emerge-Pkg to install software like gnome-desktop or qt6base / kate etc. into your system.