rxrbln/t2sde #169: Booting installation iso from usb disk results in boot error

kidharb, 2024-05-09:
I have a non legacy boot laptop and a legacy boot laptop.
I have written `t2-24.5-x86-64-base-wayland-glibc-gcc-nocona.iso` to the USB flash disk via `d bs=4M if=./t2-24.5-x86-64-base-wayland-glibc-gcc-nocona.iso of=/dev/sda status=progress oflag=sync`

The USB disk boots on the legacy boot laptop but does not boot on the non-legacy boot laptop.

On the non legacy boot laptop I get the following Grub error
`/boot/grub/x86_64-efi/normal.mod` not found

If I run isohybrid on the iso file before writing to the flash disk, I get the following error
```
isohybrid --uefi t2-24.5-x86-64-base-wayland-glibc-gcc-nocona.iso                                      
isohybrid: t2-24.5-x86-64-base-wayland-glibc-gcc-nocona.iso: boot loader does not have an isolinux.bin hybrid signature. Note that isolinux-debug.bin does not support hybrid booting
```

So `isohybrid` cannot convert the iso into hybrid mode

Am I using the correct iso or is there a step that I am missing?

coolluke1992, 2024-05-10:
is windows secure boot off on the uefi

kidharb, 2024-05-10:
Yes it is
[image]

coolluke1992, 2024-05-10:
unless you have a real reason for uefi you could get away with legacy bios mode if its got it could be called CSM i think unless its a real problem with those isos but they work with QEMU EFI vms so unless rene has a dell to test uefi with im not sure what else it could be

kidharb, 2024-05-10:
I wish I could I enable legacy bios mode, but I cannot :-(

[image]

kidharb, 2024-05-10:
I will work on creating a "hybrid" ISO image.
Can I contribute to the project here or on https://svn.exactcode.de?

coolluke1992, 2024-05-10:
put a patch into this issue by just dragging it into the box or fork the git repo and pull request or if you have the discord you can send there too

kidharb, 2024-05-10:
Please can you send me the discord invite

coolluke1992, 2024-05-10:
https://discord.gg/Ty4rqJ8j - one use link