rxrbln/t2sde #170: Manjaro Linux : Unabe to build T2 SDE /usr/bin/time not found

kidharb, 2024-05-10:
I am following the instructions from http://t2sde.org/documentation/buildintro.html

I performed step 2,3 and skipped step 4

Step 5 results in the error below

```
➜  t2-trunk sudo ./scripts/Build-Target -cfg system
Paranoia Check: No active swap partition found!
That can cause the build scripts to stall your system!. Better
activate swap using the 'swapon' command and try again.

Building tools in build/.../TOOLCHAIN/cross:
  Building cross/wrapper/cmd_wrapper
  Building cross/wrapper/strip_wrapper
  Building cross/wrapper/{arch,uname}
  Building cross/wrapper/install_wrapper
  Building cross/bin/getdu
  Building cross/bin/getfiles
  Building cross/bin/fl_wrparse
  Building cross/bin/fl_stparse
  Building cross/lib/fl_wrapper.so
  Testing cross/lib/fl_wrapper.so
  Creating cross/lib/parse-config
  Creating cross/lib/pkg_*_{pre,post}.conf
egrep: warning: egrep is obsolescent; using grep -E
INFO: Auto-detecting best mirror:
INFO: Downloading mirror-list from: t2sde.org
INFO: Testing <Germany (ExactCODE) - dl.t2sde.org (HTTP)> ... ok 541587 B/s
INFO: Testing <Germany (ExactCODE) - dl.t2sde.org (HTTPS)> ... ok 586563 B/s
INFO: Testing <Frace (Notag) - dl.notag.fr (HTTPS)> ... ok 325291 B/s
INFO: Saving mirror https://dl.t2sde.org/mirror/trunk (src/Mirror-Cache)
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E
egrep: warning: egrep is obsolescent; using grep -E

Downloading 00-dirtree:download/mirror//
Warning: Failed to open the file download/mirror//.incomplete: No such file or 
Warning: directory
curl: (23) Failure writing output to destination

ERROR: CURL Returned error: 23
INFO: Download from mirror failed, trying original URL.
 unrecognized!
scripts/Build-Pkg: line 297: /usr/bin/time: No such file or directory
  Unmounting loop mounts ...
➜  t2-trunk time
shell  0.10s user 0.11s system 0% cpu 8:04.54 total
children  6.87s user 6.60s system 2% cpu 8:04.54 total
➜  t2-trunk /usr/bin/time
zsh: no such file or directory: /usr/bin/time
```

If I change line 297 of scripts/Build-Pkg to the below

```
 297                 time -o "$root/var/adm/cache/$xpkg.tm" \
```

I get the following error 
```

Downloading 00-dirtree:download/mirror//
Warning: Failed to open the file download/mirror//.incomplete: No such file or 
Warning: directory
curl: (23) Failure writing output to destination

ERROR: CURL Returned error: 23
INFO: Download from mirror failed, trying original URL.
 unrecognized!
scripts/Build-Pkg: line 297: -o: command not found

real    0m0.000s
user    0m0.000s
sys     0m0.000s
  Unmounting loop mounts ...
```

Is there a recommended distribution that I should use to build from source?

coolluke1992, 2024-05-10:
1. use root
2.download a premade ISO from http://dl.t2sde.org/binary/2024/
3.tar the pkgs to a folder for a chroot if you dont want to install directly
4.try again without the need for sudo and make own ./scripts/Config if you wish

but would be much better off installing the ISO bare metal

kidharb, 2024-05-10:
Thanks.

I tried installing from a pre-built ISO but ran into https://github.com/rxrbln/t2sde/issues/169

kidharb, 2024-05-10:
I think I will continue trying to build my own ISO from source. That way I will learn about the build system and I will be able to create my own UEFI ISO image at the end. Thanks @coolluke1992

closed 2024-05-10 by kidharb