rxrbln/t2sde #183: T2 SDE easy desktop environment installation  (GNOME 3, Cinnamon) for x86, sparc64

isoriano1968, 2024-09-24:
Bounty: 250 €

Description:

T2 SDE though being a source based and develoepr friendly environment available on aprox. 25 different hardware architectures shall gather some improved user experience for new T2 SDE users and therefore an out of the box desktop installation with ease of use.

The task will be  to integrate a GNOME 3 desktop into a group build based approach  "./t2 install gnome3-base gnome3-extras ..." which requires besides building binaries for the target architecture an integration/extension into T2's current pkg management system. This has to be coordinated as well with T2's maintainers.

A working desktop envinroment also requires to setup and configure additional services (audio, gdm, etc.) and integration into T2's configuration tool "stone".

Target architectures for acceptance: x86, sparc64 (support on sparc64, if needed, will be provided)
GPU:  radeon support 

Happy coding.

default-vlvo, 2024-10-15:
why not gnome 47?

rxrbln, 2024-10-16:
because modern Gnome needs the latest HW acceleration and is often already slow on last gen, entry-level Intel or AMD systems. on 20y old in-order RISC CPUs it without latest GL or Vulkan acceleration it becomes a minute loading slideshow, ...

Ariyachan, 2026-11-15:
Is the €250 bounty still available through the project’s PayPal payout process? If it is, is the sparc64 testing support mentioned in the issue still available?