rxrbln/t2sde #185: Shutting down or rebooting causes a segfault (musl base wayland clang)

bramtechs, 2024-11-28:
With the iso `t2-24.8-x86-64-musl-base-wayland-musl-clang-nocona.iso`, both the installer and installed system cannot properly shutdown or reboot. A hard reset (holding the power button) is required.

This happens on both VirtualBox (host Windows 10) and on Thinkpad T570.

Shutting down after the installation:
[image]

`reboot` after booting from hard drive
[image]

`poweroff` after booting from hard drive
[image]

I'm new to source based distributions and T2 Linux. Apologies if this bug report does not follow any conventions or lacks required info.

kkmzero, 2025-01-03:
I tried to download ```*-x86-64-musl-base-wayland-musl-clang-nocona.iso``` for both T2 24.8 and 24.12 versions. Indeed, 24.8 segfaults as you described (even when you simply reboot before running install), however newest version T2 24.12 does not segfault when I run it on the same setup - VirtualBox running on top of Windows 10. I had no time to test this on real HW today.

Maybe you can try to download T2 24.12 and see whether this problem was solved.

bramtechs, 2025-01-03:
Checked the latest iso with VirtualBox and the problem has indeed been resolved. Closing the issue. 👍

closed 2025-01-03 by bramtechs

rxrbln, 2025-01-04:
Indeed supposed to be fixed with:
r72346 | rene | 2024-12-17 17:30:17 +0100 (Tue, 17 Dec 2024) | 3 lines
foxdrodd:
        * fix sysvinit segfaults w/ musl by including the header for basename