rxrbln/t2sde #186: android-tools does not compile (glibc, gcc) (includes adb)

bramtechs, 2024-11-28:
After running ```./t2 install android-tools```
the source file ```android-tools-35.0.2/vendor/adb/client/usb_libusb.cpp``` does not compile.

installed iso: ```t2-24.5-x86-64-base-wayland-glibc-gcc-nocona.iso```

```
[ 36%] Building C object vendor/CMakeFiles/libext2fs.dir/e2fsprogs/lib/ext2fs/ext_attr.c.o
[ 36%] Building C object vendor/boringssl/CMakeFiles/crypto.dir/crypto/asn1/tasn_fre.c.o
/srv/t2/src.android-tools.241128.231323.27685/android-tools-35.0.2/vendor/adb/client/usb_libusb.cpp:543:46: error: 'libusb_ssplus_usb_device_capability_descriptor' has not been declared
  543 |     static uint64_t ExtractMaxSuperSpeedPlus(libusb_ssplus_usb_device_capability_descriptor* cap) {
      |                                              ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[ 36%] Building C object vendor/boringssl/CMakeFiles/crypto.dir/crypto/asn1/tasn_new.c.o
[ 36%] Building C object vendor/boringssl/CMakeFiles/crypto.dir/crypto/asn1/tasn_typ.c.o
[ 36%] Building C object vendor/boringssl/CMakeFiles/crypto.dir/crypto/asn1/tasn_utl.c.o
[ 36%] Building C object vendor/boringssl/CMakeFiles/crypto.dir/crypto/base64/base64.c.o
[ 37%] Building C object vendor/boringssl/CMakeFiles/crypto.dir/crypto/bio/bio.c.o
[ 37%] Building C object vendor/CMakeFiles/libext2fs.dir/e2fsprogs/lib/ext2fs/extent.c.o
[ 37%] Building C object vendor/boringssl/CMakeFiles/crypto.dir/crypto/bio/bio_mem.c.o
/srv/t2/src.android-tools.241128.231323.27685/android-tools-35.0.2/vendor/adb/client/usb_libusb.cpp: In static member function 'static uint64_t LibusbConnection::ToConnectionSpeed(int)':
/srv/t2/src.android-tools.241128.231323.27685/android-tools-35.0.2/vendor/adb/client/usb_libusb.cpp:510:18: error: 'LIBUSB_SPEED_SUPER_PLUS_X2' was not declared in this scope; did you mean 'LIBUSB_SPEED_SUPER_PLUS'?
  510 |             case LIBUSB_SPEED_SUPER_PLUS_X2:
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~~
      |                  LIBUSB_SPEED_SUPER_PLUS
/srv/t2/src.android-tools.241128.231323.27685/android-tools-35.0.2/vendor/adb/client/usb_libusb.cpp: In static member function 'static uint64_t LibusbConnection::ExtractMaxSuperSpeedPlus(int*)':
/srv/t2/src.android-tools.241128.231323.27685/android-tools-35.0.2/vendor/adb/client/usb_libusb.cpp:548:38: error: request for member 'numSublinkSpeedAttributes' in '* cap', which is of non-class type 'int'
  548 |         for (uint8_t i = 0; i < cap->numSublinkSpeedAttributes; i++) {
      |                                      ^~~~~~~~~~~~~~~~~~~~~~~~~
[ 37%] Building C object vendor/CMakeFiles/libext2fs.dir/e2fsprogs/lib/ext2fs/fallocate.c.o
[ 37%] Building C object vendor/boringssl/CMakeFiles/crypto.dir/crypto/bio/connect.c.o
[ 37%] Building C object vendor/boringssl/CMakeFiles/crypto.dir/crypto/bio/errno.c.o
/srv/t2/src.android-tools.241128.231323.27685/android-tools-35.0.2/vendor/adb/client/usb_libusb.cpp:549:13: error: 'libusb_ssplus_sublink_attribute' was not declared in this scope
  549 |             libusb_ssplus_sublink_attribute* attr = &cap->sublinkSpeedAttributes[i];
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/srv/t2/src.android-tools.241128.231323.27685/android-tools-35.0.2/vendor/adb/client/usb_libusb.cpp:549:46: error: 'attr' was not declared in this scope
  549 |             libusb_ssplus_sublink_attribute* attr = &cap->sublinkSpeedAttributes[i];
      |                                              ^~~~
/srv/t2/src.android-tools.241128.231323.27685/android-tools-35.0.2/vendor/adb/client/usb_libusb.cpp:549:59: error: request for member 'sublinkSpeedAttributes' in '* cap', which is of non-class type 'int'
  549 |             libusb_ssplus_sublink_attribute* attr = &cap->sublinkSpeedAttributes[i];
      |                                                           ^~~~~~~~~~~~~~~~~~~~~~
/srv/t2/src.android-tools.241128.231323.27685/android-tools-35.0.2/vendor/adb/client/usb_libusb.cpp: In member function 'void LibusbConnection::RetrieveSpeeds()':
/srv/t2/src.android-tools.241128.231323.27685/android-tools-35.0.2/vendor/adb/client/usb_libusb.cpp:577:22: error: 'LIBUSB_BT_SUPERSPEED_PLUS_CAPABILITY' was not declared in this scope; did you mean 'LIBUSB_BT_SS_USB_DEVICE_CAPABILITY'?
  577 |                 case LIBUSB_BT_SUPERSPEED_PLUS_CAPABILITY: {
      |                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                      LIBUSB_BT_SS_USB_DEVICE_CAPABILITY
[ 37%] Building C object vendor/boringssl/CMakeFiles/crypto.dir/crypto/bio/fd.c.o
[ 37%] Building C object vendor/CMakeFiles/libext2fs.dir/e2fsprogs/lib/ext2fs/fileio.c.o
/srv/t2/src.android-tools.241128.231323.27685/android-tools-35.0.2/vendor/adb/client/usb_libusb.cpp:578:21: error: 'libusb_ssplus_usb_device_capability_descriptor' was not declared in this scope; did you mean 'libusb_ss_usb_device_capability_descriptor'?
  578 |                     libusb_ssplus_usb_device_capability_descriptor* cap = nullptr;
      |                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                     libusb_ss_usb_device_capability_descriptor
[ 37%] Building C object vendor/CMakeFiles/libext2fs.dir/e2fsprogs/lib/ext2fs/freefs.c.o
In file included from /srv/t2/src.android-tools.241128.231323.27685/android-tools-35.0.2/vendor/adb/client/file_sync_client.cpp:46:
/srv/t2/src.android-tools.241128.231323.27685/android-tools-35.0.2/vendor/adb/compression_utils.h: In member function 'virtual DecodeResult BrotliDecoder::Decode(std::span<char, 18446744073709551615>*)':
/srv/t2/src.android-tools.241128.231323.27685/android-tools-35.0.2/vendor/adb/compression_utils.h:172:5: warning: control reaches end of non-void function [-Wreturn-type]
  172 |     }
      |     ^
/srv/t2/src.android-tools.241128.231323.27685/android-tools-35.0.2/vendor/adb/client/usb_libusb.cpp:578:69: error: 'cap' was not declared in this scope
  578 |                     libusb_ssplus_usb_device_capability_descriptor* cap = nullptr;
      |                                                                     ^~~
/srv/t2/src.android-tools.241128.231323.27685/android-tools-35.0.2/vendor/adb/client/usb_libusb.cpp:579:26: error: 'libusb_get_ssplus_usb_device_capability_descriptor' was not declared in this scope; did you mean 'libusb_get_ss_usb_device_capability_descriptor'?
  579 |                     if (!libusb_get_ssplus_usb_device_capability_descriptor(
      |                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                          libusb_get_ss_usb_device_capability_descriptor
[ 37%] Building C object vendor/CMakeFiles/libext2fs.dir/e2fsprogs/lib/ext2fs/gen_bitmap64.c.o
/srv/t2/src.android-tools.241128.231323.27685/android-tools-35.0.2/vendor/adb/client/usb_libusb.cpp:582:25: error: 'libusb_free_ssplus_usb_device_capability_descriptor' was not declared in this scope; did you mean 'libusb_free_ss_usb_device_capability_descriptor'?
  582 |                         libusb_free_ssplus_usb_device_capability_descriptor(cap);
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                         libusb_free_ss_usb_device_capability_descriptor
[ 37%] Building C object vendor/boringssl/CMakeFiles/crypto.dir/crypto/bio/file.c.o

```

rxrbln, 2024-11-28:
welcome to T2 and thank you for report. Unfortunately I can reproduce it. So I guess some incompatible libusb changes. Maybe easy to patch or fixed in a new android-tools version? I can continue to try to take a quick look ...

rxrbln, 2024-11-28:
Turns out they rely on unreleased libusb API additions, I quickly imported them (hope it does not break anything;-)
Committed revision 72020.

closed 2024-11-28 by rxrbln