rxrbln/t2sde #189: Pale Moon does not build due to missing tar argument

kkmzero, 2024-12-26:
Pale Moon web browser fails to build/install with following message:
```console
15:48:58 Building 9-www/palemoon (33.5.0) ~13s
  Building in src.palemoon.241226.154857.20084
  Writing output to $root/var/adm/logs/9-palemoon.out
! Extracting Pale-Moon-33.5.0.tar.gz (-xf) ... 
! Fixing ownership and permissions ...
! Removing CVS, .svn, {arch} and .arch-ids directories ...
! Changeing into /usr/src/t2-src/src.palemoon.241226.154857.20084/pale-moon ...
! tar: option requires an argument -- 'f'
! Try 'tar --help' or 'tar --usage' for more information.
! Due to previous errors, no 9-palemoon.log file!
! (Try enabling xtrace in the config to track an error inside the build system.)
  +00:00:08 Aborted building www/palemoon
```

Any ideas how to fix this?
Result is the same for current version of Pale Moon 31.2.0 in T2 and newest version 33.5.0. If you are interested in updating this package, here's generated palemoon.desc file, however obviously I could not test whether it even works:

```text
[COPY] --- T2-COPYRIGHT-NOTE-BEGIN ---
[COPY] T2 SDE: package/*/palemoon/palemoon.desc
[COPY] Copyright (C) 2004 - 2022 The T2 SDE Project
[COPY] Copyright (C) 1998 - 2004 ROCK Linux Project
[COPY] 
[COPY] This Copyright note is generated by scripts/Create-CopyPatch,
[COPY] more information can be found in the files COPYING and README.
[COPY] 
[COPY] This program is free software; you can redistribute it and/or modify
[COPY] it under the terms of the GNU General Public License version 2.
[COPY] --- T2-COPYRIGHT-NOTE-END ---

[I] Goanna-based web browser

[T] Pale Moon is an Open Source, Goanna-based web browser.

[U] https://palemoon.org

[A] Palemoon Productions
[M] Rene Rebe <rene@t2-project.org>

[C] extra/office
[F] NOPARALLEL

[L] MPL
[S] Stable
[V] 33.5.0
[P] X -----5---9 152.700

[D] 73dbbc90559b8fe68464fbcd5633f19fad41c09968e46bef8320dedf Pale-Moon-33.5.0.tar.gz !https://repo.palemoon.org/MoonchildProductions/Pale-Moon/archive/33.5.0_Release.tar.gz
#[D] 8bb23ba1ec10fbbc024fb315b68cef10da4641fe1b42746546e80ff2 Pale-Moon-28.13.0.tar.gz https://github.com/MoonchildProductions/Pale-Moon/archive/28.13.0_Release/
#[D] 240e754524fb1457bb710a93e3396dbec5b06140e9b7e140c233ff23 UXP-RELBASE_20200901.tar.gz https://github.com/MoonchildProductions/UXP/archive/RELBASE_20200901/

```

rxrbln, 2024-12-27:
hey, i see the packages did not get the most updates since I packaged it once live on YT, and the only few updates a year ago broke the build. I updated it again and cleaned it up a bit: Committed revision 72605. But it unfortunately still needs a bit more patch love to build. Patched welcome :-)!

anrub, 2024-12-31:
@kkmzero it should work now with the latest commits to t2, you just need to enable the python2 package and install it (yes - palemoon requires python2 for building) (e.g. by adding a X to the beginning of the line with python2 in config/default/packages)

kkmzero, 2025-01-01:
> @kkmzero it should work now with the latest commits to t2, you just need to enable the python2 package and install it (yes - palemoon requires python2 for building) (e.g. by adding a X to the beginning of the line with python2 in config/default/packages)

I tried to rebuild it today after I've seen that René did a stream about Pale Moon on Twitch. It wouldn't build for me but the error was related to something other than python2, I can't remember what was the problem exactly. I can try to fetch all recent changes via SVN, do a clean rebuild tomorrow or some other day and I'll try to provide more information if necessary. If it builds then awesome, one more application/package is good to go and that's all I care about 👍 

I was somewhat of a defeatist when it comes to Linux and (F)OSS recently and I expressed my depressive opinions on René's YouTube channel (sorry) - but man, I fell in love with T2 during these holidays and I really want it to succeed. I'm currently dealing with a lot of packages that are not building, mostly cursed GUI stuff with insane amount of random unmaintained dependencies, and my hope is that we can make it work out of the box for all users on as many platforms as possible, patch by patch.

Thank you all for working on this issue (and other issues), especially during holidays. Happy New Year to you guys. ❤️ T2

rxrbln, 2025-01-01:
Amazing, happy to hear and welcome to T2! Palemoon indeed built for me w/ python2 installed. That is of course not that amazing in this day and age, but , ... Looking fwd. to more cursed abondone package patches! ;-)

closed 2025-01-01 by rxrbln