rxrbln/t2sde #190: Error while building numerous packages -  rst2man not found

kkmzero, 2025-01-05:
I run into error while building gtk/gtk+ and fun stuff like gdk-pixbuf, T2 is reporting that package rst2man is not found while checking for dependencies. I figured out that it could be part of [docutils](https://t2sde.org/packages/docutils) package already available in T2, however I am unsure. So, I installed it but I couldn't find rst2man anyway. Was this package renamed? Does it need to be packaged separately if we can verify that it's part of docutils? Is there something wrong with configuration of docutils or am I doing something wrong?

foxdrodd, 2025-01-05:
Looks like rst2man is indeed part of docutils but a python script "rst2man.py", you can also grep through the files in /var/adm/flists for installed package files.  Does that already fixed your gtk build issue, or is there more?

kkmzero, 2025-01-05:
It fails to build of course but for other reasons (I guess?). I reached a point in which I don't want to touch these packages ever again. If we confirmed that rst2man is part of docutils, would change like this make sense  though?

```patch
Index: package/gnome/gdk-pixbuf/gdk-pixbuf.cache
===================================================================
--- package/gnome/gdk-pixbuf/gdk-pixbuf.cache	(revision 72768)
+++ package/gnome/gdk-pixbuf/gdk-pixbuf.cache	(working copy)
@@ -24,7 +24,7 @@
 [DEP] openssl
 [DEP] pkgconfig
 [DEP] python
-[OPT] rst2man
+[DEP] docutils
 [DEP] sed
 [DEP] tar
 [DEP] xz

```

foxdrodd, 2025-01-05:
For gdk-pixbuf, docutils/rst2man is optional, so it should not bother the success of the built - I updated the dependency to docutils instead of rst2man.

Feel free to share your error log, maybe someone can help with it (or fix a bug). Desktop related packages are unfortunately always  a real blessing to build..

kkmzero, 2025-01-05:
Well, pretty much 80% of these desktop packages won't build for me. Even after following instructions you run into errors, just this command from T2 KB will produce some nice errors:

```console
./t2 install -f gobject-introspection gdk-pixbuf pango harfbuzz at-spi2-core \
graphene gtk
```

Some of my results:
```console
20:37:51 Building 9-gnome/gdk-pixbuf (2.42.12) ~39s
  Building in src.gdk-pixbuf.250105.203751.10832
  Creating backup of old package data.
  Writing output to $root/var/adm/logs/9-gdk-pixbuf.out
! [29/161] Compiling C object gdk-pixbuf/pixops/libpixops.a.p/pixops.c.o
! [30/161] Generating gdk-pixbuf/GdkPixbuf-2.0.gir with a custom command (wrap
! FAILED: gdk-pixbuf/GdkPixbuf-2.0.gir 
! env PKG_CONFIG_PATH=/lib64/pkgconfig:/lib/pkgconfig:/usr/share/pkgconfig:/us
! Couldn't find include 'GModule-2.0.gir' (search path: '['/opt/gnome/share/gi
! ninja: build stopped: subcommand failed.
! Due to previous errors, no 9-gdk-pixbuf.log file!
! (Try enabling xtrace in the config to track an error inside the build system
  +00:00:41 Aborted building gnome/gdk-pixbuf

20:42:11 Building 9-gnome/gtk (4.16.7) ~21m:51s
  Building in src.gtk.250105.204211.29873
  Writing output to $root/var/adm/logs/9-gtk.out
bash-5.2# ! Found CMake: /usr/bin/cmake (3.31.3)
! Run-time dependency gdk-pixbuf-2.0 found: NO (tried pkgconfig and cmake)
! meson.build:415:20: ERROR: Dependency "gdk-pixbuf-2.0" not found, tried  ..
! A full log can be found at /usr/src/t2-src/src.gtk.250105.204211.29873/g ..
! Due to previous errors, no 9-gtk.log file!
! (Try enabling xtrace in the config to track an error inside the build sy ..
  +00:00:24 Aborted building gnome/gtk
```

Harfbuzz once installed for me, doing ./t2 install -f harfbuzz suddenly failed to re-build it again, so it made fail this one:
```console
20:38:33 Building 9-gnome/pango (1.54.0) ~1m:35s
  Building in src.pango.250105.203833.15903
  Creating backup of old package data.
  Writing output to $root/var/adm/logs/9-pango.out
! [48/160] Compiling C object pango/libpangoft2-1.0.so.0.5400.0.p/pangoft2.c.o
! [49/160] Generating pango/Pango-1.0.gir with a custom command (wrapped by me
! FAILED: pango/Pango-1.0.gir 
! env PKG_CONFIG_PATH=/lib64/pkgconfig:/lib/pkgconfig:/usr/share/pkgconfig:/us
! Couldn't find include 'HarfBuzz-0.0.gir' (search path: '['/opt/gnome/share/g
! ninja: build stopped: subcommand failed.
! Due to previous errors, no 9-pango.log file!
! (Try enabling xtrace in the config to track an error inside the build system
! opt/gnome/lib64/libpangoxft-1.0.so.0: pango pangomm
! opt/gnome/lib64/pkgconfig/pango.pc: pango pangomm
! opt/gnome/lib64/pkgconfig/pangocairo.pc: pango pangomm
! opt/gnome/lib64/pkgconfig/pangofc.pc: pango pangomm
! opt/gnome/lib64/pkgconfig/pangoft2.pc: pango pangomm
! opt/gnome/lib64/pkgconfig/pangoot.pc: pango pangomm
! opt/gnome/lib64/pkgconfig/pangoxft.pc: pango pangomm
! Due to previous errors.
```

2 days ago I did clean install of T2 minimal without Xorg. Certain Xorg components wouldn't build, so I re-installed it with minimal install+Xorg and now I have to deal with this stuff? Man... patches welcome, I guess...

By the way, I don't know if it's still the case but after Xfce update to 4.20 there are still packages that won't build precisely because of this cursed Gobject GModule cursed GIR stuff or whatever it even is. Gobject-introspection builds but it doesn't seem to matter. Luckily I don't need desktop environments on my servers but it must suck for people who just want to do ```./t2 install i3``` or ```./t2 install gimp``` and the command line is spitting out errors left and right.

foxdrodd, 2025-01-05:
Can you: -f install ` gobject-introspection`  and then ` glib`  . then again the first command `./t2 install -f gobject-introspection gdk-pixbuf pango harfbuzz at-spi2-core graphene gtk`

kkmzero, 2025-01-05:
**Note: There's apparently ```gstreamer-player``` package missing (Warning: No such package.) set as dependency for gtk, I noticed it just right now while running third command.**

Results:
1. -f install of ```gobject-introspection``` succeeded as before
2. -f install of ```glib``` succeeded
3. -f install of those remaining packages now mostly works, however there is still one error:

```console
21:32:36 Building 9-gnome/pango (1.54.0) ~1m:35s
  Building in src.pango.250105.213235.19244
  Creating backup of old package data.
  Writing output to $root/var/adm/logs/9-pango.out
! opt/gnome/lib64/libpangoxft-1.0.so.0: pango pangomm
! opt/gnome/lib64/pkgconfig/pango.pc: pango pangomm
! opt/gnome/lib64/pkgconfig/pangocairo.pc: pango pangomm
! opt/gnome/lib64/pkgconfig/pangofc.pc: pango pangomm
! opt/gnome/lib64/pkgconfig/pangoft2.pc: pango pangomm
! opt/gnome/lib64/pkgconfig/pangoot.pc: pango pangomm
! opt/gnome/lib64/pkgconfig/pangoxft.pc: pango pangomm
! Due to previous errors.
  +00:01:32 Aborted building gnome/pango
```

It seems that (re)installing ```glib``` is a solution but afaik it's not mentioned in documentation and not one package depending on it notified me about glib missing or anything. How many packages are in T2 that could fail because of this issue? Basically most DEs/WMs and all GTK apps? So I don't know what could help users the most - adding glib to hundreds of cache files and/or adding this information to T2 Knowledge Base, or something else. Or am I just too stupid to use T2?

foxdrodd, 2025-01-06:
That pango/mm error is because of shared files between pango and pangomm, though it is not a build error but more of a warning while installing the files - so pango is successfully installed, but you get this message about potentially conflicting packages, as they clobber their files.

Regarding the whole gobject-introspection/Gir issue from https://t2sde.org/kb/8/ : 
> gobject-introspection does not support cross-compilation as it requires to run native helper binaries to process the annoated source code for introspection. 

So the common first thing to do is to rebuild gobject-introspection, glib and other Gir-related stuff. After that, desktop/gnome packages can build.

Thanks for noticing the orphan dependency, will check them further, though it won't break the build.

kkmzero, 2025-01-06:
Thank you very much for helping me and for giving me useful information, I really appreciate it.

When it comes to glib etc., I obviously didn't know it's a _common_ thing to do. I'm still of the opinion that it would be nice if the documentation would be just a little bit more verbose and tell me to build gobject-introspection, then glib, then (re)build all of those 7 packages and then go from there or whatever the correct process is.

If you follow the documentation now, it tells you that after running that one command, you should have .gir "gobject introspection data" generated for other packages but that's not true, at least not on minimal system installation. Imho this piece of information could save a lot of time in the future.