rxrbln/t2sde #192: Bluez 5.72 broken

NAHTAIV3L, 2025-01-07:
After pairing some bluetooth devices (I tested with bose headphones and airpods) restarting bluetoothd causes you to have to re-pair the devices. Otherwise it works correctly.
I believe that this has alread been fixed upstream (I found an issue on the bluez github that was the exact issue and it was closed because of a patch). 
The package cannot be updated because it requires a newer version of udev which is now part of systemd.

rxrbln, 2025-01-07:
Uff, I don't remember such an issue the last time I used bluetooth, but I rarely do and have just working wired devices ;-)
I had worked on updating udev, building it out of systemd. I had it building, but did not fully work at run-time. The tarball of that WIP is on our discord already. I can try to debug it the coming days to finally update udev to a modern for important updates like this.

NAHTAIV3L, 2025-01-08:
Before being on t2sde I was on gentoo and the older version that they had did not have this bug so it is a regression which is probably why you didn't have this issue before. I tried doing what gentoo does and just building udev from systemd and installing it but it didn't work and is not a great solution

rxrbln, 2025-01-09:
> Before being on t2sde I was on gentoo and the older version that they had did not have this bug so it is a regression which is probably why you didn't have this issue before. I tried doing what [...]

I now committed bluez updates w/ patching using the newer udev hwdb api away. Does this updated bluez work better for you? I'll keep working on updating udev to the latest out of systemd, sigh.

NAHTAIV3L, 2025-01-09:
I tried this and it was broken the same way as before. There isn't much to do but deal with udev I'm just going to repair things for now until udev is build out of systemd thank you for trying.

rxrbln, 2025-01-10:
thank you for the report. I knew we finally need to deal with modern u/dev updates for quite some years already and had tried it at times. I finally got the latest udev out of systemd to actually work yesterday and committed that update, too. There are still some minor details to cleanup and commit. When those are committed, one needs to update the initrd and system init script as udevadm trigger now needs to run with --action=add for the new udev rules to actually load the driver modules. I think we can leave this issue open until Bluetooths works flawlessly again. I will try to use some bluetooth devices the coming days, too.

kkmzero, 2025-01-10:
Recent update of 00-dirtree, mkinitrd and udev built but it broke for me something with the resolution of CLI and i3wm will start via xinit but immediately hangs. I don't know if other DEs still work. Do I need to also rebuild Linux kernel or some other packages?

kkmzero, 2025-01-10:
> thank you for the report. I knew we finally need to deal with modern u/dev updates for quite some years already and had tried it at times. I finally got the latest udev out of systemd to actually [...]

Which one is the system init script?

NAHTAIV3L, 2025-01-10:
I tried the updated udev and bluez packages and they work as they should. This can now be closed

NAHTAIV3L, 2025-01-10:
You have to update every package that depends on udev and also update sysvinit

closed 2026-01-04 by foxdrodd