rxrbln/t2sde #193: Apache fails to build

kkmzero, 2025-01-10:
```console
[35;1m19:07:03 Building 9-network/apache (2.4.62) ~4m:38s[0m
pkgprefix: package bdb is not present
pkgprefix: package bdb is not present
  Building in src.apache.250110.190703.24143[0m
  Creating backup of old package data.[0m
  Writing output to $root/var/adm/logs/9-apache.out[0m
[31;1m![0m Making all in srclib
[31;1m![0m Making all in os
[31;1m![0m Making all in unix
[31;1m![0m /usr/share/build-1/libtool --silent --mode=compile gcc  -g -O2 -Wall      -DLINUX -D_REENTRANT -D_GNU_SOURCE     -I. -I/usr/src/t2-src/src.apache.250110.1 ..
[31;1m![0m /usr/share/build-1/libtool --silent --mode=link gcc  -g -O2 -Wall      -L/srv/t2/build/x86-64-24.12-generic-x86-64-nocona-cross-linuxlib64 -L/srv/t2/build ..
[31;1m![0m /usr/share/build-1/libtool: x86_64-t2-linux-gnu-ar: inaccessible or not found
[31;1m![0m Due to previous errors, no 9-apache.log file!
[31;1m![0m (Try enabling xtrace in the config to track an error inside the build system.)
  [31;1m+00:00:18 Aborted building network/apache[0m
```

```console
 /usr/share/build-1/libtool: x86_64-t2-linux-gnu-ar: inaccessible or not found
```

foxdrodd, 2025-01-11:
Could you try rebuilding apr `t2 install -f apr` and then apache again, not sure yet, why it happens in the first place.

rxrbln, 2025-01-11:
maybe cross build related? libtool?

kkmzero, 2025-01-11:
> Could you try rebuilding apr `t2 install -f apr` and then apache again, not sure yet, why it happens in the first place.

This solved it for some reason, I have no idea why it wouldn't build the first time. Thank you both for your replies and for helping me to solve this issue.

rxrbln, 2025-01-11:
Most likely some file does include residual base ISO cross compiled information that do not work for native builds. I had seen this before and we should fix the pkg cross build (libtoo or apr-*) to not leak cross compile details.