rxrbln/t2sde #194: texlive is outdated

NAHTAIV3L, 2025-01-12:
texlive is currently on the 2009 release and fails to build. 
I tried updating it to the 2024 version and it built but I could not get it to run.

rxrbln, 2025-01-12:
would you have the patch of what you tried to update so far? Thanks!

NAHTAIV3L, 2025-01-12:
here is what I got to compile it doesnt use makeopt which is probably not good but that is what was there before.
[texlive.patch](https://github.com/user-attachments/files/18390976/texlive.patch)

rxrbln, 2025-01-12:
m4tic sent a patch today, too: Committed revision 72860. Thanks!

closed 2025-01-12 by rxrbln

NAHTAIV3L, 2025-01-13:
The package builds but it still does not work. This is what happens when trying to run latex command.
```
[nahtaiv3l@T14nahtaiv3l latex]$ latex test.tex
This is pdfTeX, Version 3.141592653-2.6-1.40.26 (TeX Live 2024) (preloaded format=latex)
 restricted \write18 enabled.

kpathsea: Running mktexfmt latex.fmt
Can't locate mktexlsr.pl in @INC (@INC entries checked: /opt/tlpkg /opt/texmf-dist/scripts/texlive /usr/lib64/perl5/site_perl/5.38.0/x86_64-t2-linux-gnu /usr/lib64/perl5/site_perl/5.38.0 /opt/lib64/perl5/vendor_perl/5.38.0/x86_64-t2-linux-gnu /opt/lib64/perl5/vendor_perl/5.38.0 /usr/lib64/perl5/5.38.0/x86_64-t2-linux-gnu /usr/lib64/perl5/5.38.0) at /opt/texlive/bin/mktexfmt line 23.
BEGIN failed--compilation aborted at /opt/texlive/bin/mktexfmt line 25.
I can't find the format file `latex.fmt'!
```

scsijon, 2025-01-13:
Try what's in this> https://latex.org/forum/viewtopic.php?t=2023 <it's 
happened before and will again no doubt, use to be in the build 
instructions to test with, don't know where it's gone.

On 13/1/25 13:13, NAHTAIV3L wrote:
> The package builds but it still does not work. This is what happens when trying to run latex command.
> ```
> ***@***.*** latex]$ latex test.tex
> This is pdfTeX, Version 3.141592653-2.6-1.40.26 (TeX Live 2024) (preloaded format=latex)
>   restricted \write18 enabled.
> 
> kpathsea: Running mktexfmt latex.fmt
> Can't locate mktexlsr.pl in @INC ***@***.*** entries checked: /opt/tlpkg /opt/texmf-dist/scripts/texlive /usr/lib64/perl5/site_perl/5.38.0/x86_64-t2-linux-gnu /usr/lib64/perl5/site_perl/5.38.0 /opt/lib64/perl5/vendor_perl/5.38.0/x86_64-t2-linux-gnu /opt/lib64/perl5/vendor_perl/5.38.0 /usr/lib64/perl5/5.38.0/x86_64-t2-linux-gnu /usr/lib64/perl5/5.38.0) at /opt/texlive/bin/mktexfmt line 23.
> BEGIN failed--compilation aborted at /opt/texlive/bin/mktexfmt line 25.
> I can't find the format file `latex.fmt'!
> ```
>