rxrbln/t2sde #206: exrtools fails to build

kkmzero, 2025-02-12:
Package exrtools fails to build. Likewise, its dependency ilmbase does not build, probably needs patch for new compilers and I don't know how to solve that (yet):

**ilmbase:**
```console
  Building in src.ilmbase.250212.102002.6700
  Writing output to $root/var/adm/logs/9-ilmbase.out
! ImathVec.h:2109:26: error: ISO C++17 does not allow dynamic exception specifications
!  2109 | Vec4<T>::normalizeExc () throw (IEX_NAMESPACE::MathExc)
!       |                          ^~~~~
! ImathVec.h:2150:33: error: ISO C++17 does not allow dynamic exception specifications
!  2150 | Vec4<T>::normalizedExc () const throw (IEX_NAMESPACE::MathExc)
!       |                                 ^~~~~
! Due to previous errors, no 9-ilmbase.log file!
! (Try enabling xtrace in the config to track an error inside the build system.)
  +00:00:26 Aborted building graphic/ilmbase
```

**exrtools:**
```console
  Building in src.exrtools.250212.102831.23876
  Writing output to $root/var/adm/logs/9-exrtools.out
! pngoutput.c: In function 'pngoutput_new':
! pngoutput.c:93:56: error: 'Z_BEST_COMPRESSION' undeclared (first use in this function); did you mean 'PN ..
!    93 |         png_set_compression_level( pngoutput->png_ptr, Z_BEST_COMPRESSION );
!       |                                                        ^~~~~~~~~~~~~~~~~~
!       |                                                        PNG_Z_DEFAULT_COMPRESSION
! pngoutput.c:93:56: note: each undeclared identifier is reported only once for each function it appears in
! Due to previous errors, no 9-exrtools.log file!
! (Try enabling xtrace in the config to track an error inside the build system.)
  +00:00:22 Aborted building graphic/exrtools
```

kkmzero, 2025-02-12:
No, I do not plan to use it at all, I was just testing pkgs.

rxrbln, 2025-02-12:
Thanks for bring this to my attention, it turned out our ilmbase way outdated.
It was an easy hot fix after that, just including zlib.h.
Committed revision 74149.

closed 2025-02-12 by rxrbln