rxrbln/t2sde #224: SDL situation in T2

kkmzero, 2025-03-20:
Since I'm doing some SDL development recently and I'm using T2 for testing, I was interested in updating SDL library to newer version 3.x.x. I realized that SDL packages in T2 are kinda weird and adding libsdl3 pkg wouldn't be the most ideal solution imho. I would like to gather some information and track the issues before I proceed.

Some observations:
- T2 has available both libsdl_ttf and libsdl2_ttf - both are slightly different version 2, so I do not know what is going on there
- There is some tiny chance that we don't even need to distribute those old SDL1 packages anymore, I hadn't got time to check DEPs of other packages yet
- Older versions of SDL (libsdl) can MAYBE be replaced with [sdl12-compat](https://github.com/libsdl-org/sdl12-compat) and/or [sdl2-compat](https://github.com/libsdl-org/sdl2-compat) and SDL3 would become just libsdl in T2
- Download links could link to GitHub and then Data could maybe take care of future updates more reliably?
- Some packages might be unavailable, for example libsdl_perl(?)
- I don't know YET whether installing both SDL1 and SDL2 leads to conflicts or if all of those pkgs are properly configured

I am willing to put the work in and try to see whether these packages even build/install and whether we need to keep them around for old software compatibility reasons but I wanted to check first whether this is something that needs attention or if this is how it's supposed to be and there's nothing wrong.

kkmzero, 2025-03-20:
Some messed up stuff I noticed:
- [libsdl2_gfx](https://t2sde.org/packages/libsdl2_gfx) is version 1.0.4 and [libsdl_gfx](https://t2sde.org/packages/libsdl_gfx) is version 2.0.19beta, so reveresed branches?
- [libsdl_gfx](https://t2sde.org/packages/libsdl_gfx) doesn't even download for me anymore, so we have only version 1.0.4 available now?

Other notes, please check if valid:
- [libsdl_perl](https://t2sde.org/packages/libsdl_perl) is needed only for [frozen-bubble](https://t2sde.org/packages/frozen-bubble), download throws out file type mismatch error
- [libsdl_pango](https://t2sde.org/packages/libsdl_pango) is only needed for [frozen-bubble](https://t2sde.org/packages/frozen-bubble)
- [libsdl_sound](https://t2sde.org/packages/libsdl_sound) is needed only for [asc](https://t2sde.org/packages/asc) and [gltron](https://t2sde.org/packages/gltron)

kkmzero, 2025-03-20:
Another thing, there is [sdl2_pango](https://t2sde.org/packages/sdl2_pango) which is not named libsdl2_pango and it seems no other package depends on it, please check if valid.

I also checked, T2 has [sdl12-compat](https://t2sde.org/packages/sdl12-compat) but not sdl2-compat YET, so that should be easy to add, however the question is if it is working replacement for libsdl and libsdl2 pkgs.

rxrbln, 2025-03-20:
thank you for the review and report. I took a quick look and it appers the ttf and gfx packages are the right matches but just have strange upstream versions unrelated of the major SDL version. I found a new libsdl_gfx and updated it, too. The sdl2_pango could certainly be renamed, too. Even if not yet used by a t2 pkg someone probably needed it, so no need to delete until another t2 pkg will use it.

kkmzero, 2025-03-20:
Thank you very much for looking into this. Here's libsdl3 based on libsdl2 desc. It built, created dir in /usr/include, please check if it's okay, I have no idea.
```patch
Index: package/graphic/libsdl3/libsdl3.cache
===================================================================
--- package/graphic/libsdl3/libsdl3.cache	(nonexistent)
+++ package/graphic/libsdl3/libsdl3.cache	(working copy)
@@ -0,0 +1,45 @@
+[TIMESTAMP] 1742489439 Thu Mar 20 17:50:39 2025
+[BUILDTIME] 25 (9)
+[SIZE] 6.13 MB, 106 files
+
+[DEP] alsa-lib
+[DEP] bash
+[DEP] binutils
+[DEP] cmake
+[DEP] coreutils
+[OPT] dbus
+[DEP] diffutils
+[OPT] esound
+[DEP] freetype
+[DEP] gawk
+[DEP] grep
+[DEP] gzip
+[OPT] jack
+[OPT] jack2
+[OPT] libdecor
+[DEP] libdrm
+[DEP] libglvnd
+[OPT] libsamplerate
+[DEP] libusb
+[OPT] libx11
+[OPT] libxcb
+[OPT] libxcursor
+[OPT] libxext
+[OPT] libxfixes
+[OPT] libxi
+[OPT] libxrandr
+[OPT] libxrender
+[OPT] libxscrnsaver
+[DEP] linux-header
+[DEP] make
+[DEP] mesa
+[OPT] pipewire
+[DEP] pkgconfig
+[OPT] pulseaudio
+[DEP] sed
+[DEP] tar
+[DEP] udev
+[DEP] unwind
+[OPT] wayland
+[OPT] xkbcommon
+[DEP] xorgproto
Index: package/graphic/libsdl3/libsdl3.desc
===================================================================
--- package/graphic/libsdl3/libsdl3.desc	(nonexistent)
+++ package/graphic/libsdl3/libsdl3.desc	(working copy)
@@ -0,0 +1,46 @@
+[COPY] --- T2-COPYRIGHT-BEGIN ---
+[COPY] t2/package/*/libsdl3/libsdl3.desc
+[COPY] Copyright (C) 2025 The T2 SDE Project
+[COPY] SPDX-License-Identifier: GPL-2.0
+[COPY] --- T2-COPYRIGHT-END ---
+
+[I] A cross-platform Graphic API
+
+[T] Simple DirectMedia Layer is a cross-platform multimedia library
+[T] designed to provide fast access to the graphics framebuffer and
+[T] audio device. It is used by MPEG playback software, emulators, and
+[T] many popular games, including the award winning Linux port of
+[T] "Civilization: Call To Power." Simple DirectMedia Layer supports
+[T] Linux, Win32, BeOS, MacOS, Solaris, IRIX, and FreeBSD.
+[T]
+[T] SDL is written in C, but works with C++ natively, and has bindings
+[T] to several other languages, including Ada, Eiffel, Java, ML, Perl,
+[T] PHP, Pike, Python, and Ruby.
+
+[U] https://www.libsdl.org/
+
+[A] Sam Lantinga <slouken@libsdl.org>
+[M] Rene Rebe <rene@t2-project.org>
+
+[C] base/x11
+[F] CROSS
+
+[E] opt dbus
+[E] opt libdecor
+[E] opt libsamplerate
+[E] opt esound
+[E] opt libx11 libxcb libxcursor libxext libxfixes libxi libxrandr libxrender libxscrnsaver xkbcommon
+[E] opt wayland
+[E] opt jack
+[E] opt jack2
+[E] opt pipewire
+[E] opt pulseaudio
+
+[L] GPL
+[S] Stable
+[V] 3.2.8
+[P] X -----5---9 130.000
+
+[O] pkginstalled libx11 && var_append GCC_WRAPPER_INSERT ' ' "-L$root$(pkgprefix libdir libx11)"
+
+[D] f0790ed4d6b6660b37d12b05f9c125b56fbfb960861488d682079c0c SDL3-3.2.8.tar.gz http://www.libsdl.org/release/

```

rxrbln, 2025-03-20:
Thank you for bringing it up and paging v3:
Committed revision 75273.

rxrbln, 2025-03-21:
thanks, looks good w/ just a title punctuation removed