rxrbln/t2sde #225: 'linux-firmware' should be build before 'linux', because /sbin/mkinitrd  requires /lib/firmware/amd-ucode/microcode_amd*.bin

hpaluch, 2025-03-22:
While doing simple cross-build for x86_64 -> x86_64  (I have my own template called [15-cli.in](https://github.com/hpaluch/t2sde-files/blob/master/target/generic/pkgsel/15-cli.in) which is basically "base" without bloat (llvm, clang, rustc,cargo) so I can quickly build ISO, I found this bug:

The problem is that at stage 2:
* `linux` package calls `/sbin/mkinitrd` (from `mkinitrd` package) on install phase (that's OK)
* but `/sbin/mkinitrd` script requires `/lib/firmware/amd-ucode/microcode_amd*.bin` from `linux-firmware` package (that's problem)
* but it has wrong priority - being build *after* `linux` package (which is too late).

Here is detailed error message when cross-building `linux` in stage 2:
```
16:33:41 191/197 Building 2-kernel/linux (6.13.7) ~18m:56s                    
  Stack-Smashing Protector disabled for this package                                                                                                                  
  Position-Independent Executables disabled for this package                 
  Building in src.linux.crosscli.250322.163341.26128, with 9 threads                                                                                                  
  Writing output to $root/var/adm/logs/2-linux.out
!       /usr/sbin/smartctl NEEDS /usr/lib64/libstdc++.so.6       
!       /usr/lib64/libstdc++.so.6 SYMLINKS to libstdc++.so.6.0.33            
! '/usr/src/t2-src/src.linux.crosscli.250322.163341.26128/tmp/tmp.Cp2TKXrIQe.d/usr/lib64/libstdc++.so.6' -> 'libstdc++.so.6.0.33'
! Warning: Skipped optional file /usr/sbin/cache_check!                                                                                                               
! Warning: Skipped optional file /usr/embutils/dmesg!    
! cat: '/usr/src/t2-src/build/crosscli-25-svn-generic-x86-64-nocona-cross-linux/lib/firmware/amd-ucode/microcode_amd*.bin': No such file or directory
! Due to previous errors, no 2-linux.log file!                               
! (Try enabling xtrace in the config to track an error inside the build system.)
  +00:38:21 Aborted building kernel/linux                
```

NOTE: Above error message comes from `/sbin/mkinitrd` which is called from `package/kernel/linux/linux.conf` when
`/sbin/mkinitrd` is available (which is, because it has lower  priority number than kernel).

And here are revealed stages and priorities of my packages (I have `-cfg crosscli`):
```shell
$ grep -E ' (linux|linux-firmware|mkinitrd) ' config/crosscli/packages 
X --2------- 100.000 base mkinitrd 2025-01-10 / base/system CROSS 0
X --2------- 800.000 kernel linux 6.13.7 / base/kernel CROSS KERNEL CUSTOM-LTO NO-PIE NO-SSP 0
X -12------- 999.000 firmware linux-firmware 20250311 / base/firmware CROSS 0
```

Solution: I suggest lowering priority of `linux-firmware` to be build before `linux` (kernel). *Disclaimer: I have not enough power and/or resources to verify that this change will not break something else.*

Workaround: (stage 2 cross-build my config is called `crosscli`):
```shell
# workaround: when above errors occurs, run:
t2 build-target -cfg crosscli 2-linux-firmware
# if above builds fail with error: Created file outside basedir: /root/.parallel
# simply run it again (this time will above files already exist not triggering error)
t2 build-target -cfg crosscli 2-linux-firmware
# resume cross-build:
t2 build-target -cfg crosscli
```

FYI: here are my non-default values used for my `15-cli.in` template:
```shell
SDECFG_PKGSEL_TMPL='cli'
SDECFG_X8664_OPT='nocona'
SDECFG_CROSSBUILD='1'
SDECFG_CONTINUE_ON_ERROR_AFTER='9
```
You can find detailed info/files on that template in README file at: https://github.com/hpaluch/t2sde-files/tree/master?tab=readme-ov-file#my-first-iso-cross-build---cli-template

hpaluch, 2025-03-22:
Addendum: I have found worrying crashes in `build/crosscli-25-svn-generic-x86-64-nocona-cross-linux/var/adm/logs/1-linux-firmware.log`:
```
Running make CC=x86_64-t2-linux-gnu-gcc CPP=x86_64-t2-linux-gnu-cpp CXX=x86_64-t2-linux-gnu-g++ prefix=/usr PREFIX=/usr docdir=/usr/doc/linux-firmware CC_FOR_BUILD=cc BUILDCC=cc BUILD_CC=cc HOSTCC=cc HOST_CC=cc STRIP=x86_64-t2-linux-gnu-strip AR=x86_64-t2-linux-gnu-ar LD=x86_64-t2-linux-gnu-ld AS=x86_64-t2-linux-gnu-as RANLIB=x86_64-t2-linux-gnu-ranlib NM=x86_64-t2-linux-gnu-nm DESTDIR=/usr/src/t2-src/build/crosscli-25-svn-generic-x86-64-nocona-cross-linux install -j9
install -d /usr/src/t2-src/build/crosscli-25-svn-generic-x86-64-nocona-cross-linux/lib/firmware
./copy-firmware.sh -j9 /usr/src/t2-src/build/crosscli-25-svn-generic-x86-64-nocona-cross-linux/lib/firmware
fl_wrapper.so: write outside basedir (/usr/src/t2-src): /root/.parallel/tmp/sshlogin/t2-clean/linelen26589
Aborted (core dumped) 
fl_wrapper.so: write outside basedir (/usr/src/t2-src): /root/.parallel/tmp/sshlogin/t2-clean/linelen27669
Aborted (core dumped) 
Now run "make dedup" to de-duplicate any firmware files
Creating file list and doing final adaptions ... 
Searching for orphaned files ...
Found 5 files for this package.
Found 16 orphaned files for this package.
Calculating package dependencies ...
Deleting dependencies, pattern:  -e ".*: t2-src$" -e ".*: glibc$" -e ".*: gcc$" ...
Adding dependencies
Marking optional dependencies
Creating md5sum files ...
Creating package description ...
Making post-install adaptions.
```

But build process thinks that `1-linux-firmware.log` finished without error...

The problem is when there is passed `-jX` make arg the script `./copy-firmware.sh` (which is always but with 1 CPU core systems - quite unlikely today) will run GNU parallel, but it creates working files under `$USER/.parallel/...`

It should be easily reproducible - just `rm -rf /root/.parallel` before building `linux-firmware`.

rxrbln, 2025-03-23:
I had linux-firmware built twice, in stage 1 for cross builds, and stage 2 after the linux kernel to optimize only installing files the linux kernel actually uses to cut down the ISO size by some hundred  MB IIRC. Need to look at your report in more detailed another day, ... ;-)

hpaluch, 2025-03-23:
Actual bug was quiet failure of `1-linux-firmware` due GNU parallel behavior introduced by upstream.

I verified that workaround from https://github.com/rxrbln/t2sde/pull/226#issuecomment-2746141706,  SVN commit:
```
r75347 | rene | 2025-03-23 11:40:20 +0100 (Sun, 23 Mar 2025) | 2 lines

        * mark linux-firmware NOPARALLEL to not newly depend on gnu parallel
```
Fixes this issue. Closing.

closed 2025-03-23 by hpaluch