rxrbln/t2sde #228: Hardware acceleration on the PS3

AyoubCoding21, 2025-03-27:
Have you got hardwrae acceleration on the t2sde ? bc i cant find any links to get my ps3 in linux to have hardware acceleration.

AyoubCoding21, 2025-03-28:
@rxrbln

rxrbln, 2025-03-29:
Only proof of concept. The GPU is sometimes throwing an exception or crashing and I did not yet really have time to further debug this. I had hoped the YT content results in subscribers and income faster to finance it or others to send patches too. sooner than later we will also continue with the RSX again http://patreon.com/renerebe ;-)

AyoubCoding21, 2025-03-29:
Is it in the os (when I install it I have gpu acceleration)?

Ps : I could also help with the Linux, I'm not a expert, but I can help with testing and usermode stuff

AyoubCoding21, 2025-05-16:
Btw im wondering, if there is cfw that allows lv1 patches (like rebug cfw btw it has restored otheros and renamed otheros++ so you can update the ps3 section on your website), can we like take full advantage of the RSX ?

AyoubCoding21, 2025-05-17:
Nice