rxrbln/t2sde #234: sparc64 - segmentation fault

tankf33der, 2025-04-13:
SPARC64 booted up in `qemu 9.1.2` but svn cannot checkout:
```
# svn co http://svn.exactcode.de/t2/trunk t2-trunk
Segmentation fault
```
Tried 24.12, 24.8, 23.6.

tankf33der, 2025-04-13:
```
root@t2:~/t2-trunk# ./t2 install -deps=none git
Using /etc/SDE-CONFIG/ as default config
'/etc/SDE-CONFIG/config' -> 'config/default/config'
'/etc/SDE-CONFIG/confopt' -> 'config/default/confopt'
'/etc/SDE-CONFIG/packages' -> 'config/default/packages'
T2 25-svn configuration
Aquiring config lock ...
Building src/bash/luabash
  C         /root/t2-trunk/src/lua/lapi.o
cc: internal compiler error: Segmentation fault signal terminated program cc1
Please submit a full bug report, with preprocessed source (by using -freport-bug).
See <https://gcc.gnu.org/bugs/> for instructions.
make: *** [build/bottom.make:42: /root/t2-trunk/src/lua/lapi.o] Error 4
Error building the Lua bash accelerator
git: Not selected.
No package scheduled to build.
root@t2:~/t2-trunk# 
root@t2:~/t2-trunk# 
root@t2:~/t2-trunk# ./t2 install -deps=none git
git: Not selected.
No package scheduled to build.
root@t2:~/t2-trunk# 

```

rxrbln, 2025-04-13:
Thank you for your report. I don't quite remember that from testing on actual HW, which ISO precisely is this?

rxrbln, 2025-04-13:
It appears to only crash in Qemu. I did not test the older download, but with my latest build the svn binary indeed segfaults qemu user mode sparc64 emulation but works  just fine on my Ultra 25. I don't think I had svn segfaulting on any real SPARC hardware when I excessively tested and worked on SPARC things last year ;-/

tankf33der, 2025-04-13:
Good to know T2 is safe!!!

tankf33der, 2025-04-13:
I am porting [vlang](https://vlang.io/) language to sparc64.

I have a cheeky question - 
I am looking for access to real `SPARC` hardware, if you can let me in for T2 for a couple of weeks, I'd appreciate it. 

```
My small requirements and 100% warranty - 
o) T2 Linux
o) no cpu load
o) no CI
o) only compilation attempts with gcc+boehm-gc
o) no sh*t.
o) you can disable access later

Thanks in advance,
```

p.s. i will this close this issue in several hours.

rxrbln, 2025-04-13:
I currently don't really have the infrastructure to allow remote access or have random old servers running 24/7 for a week or two in our small office. You can leave this bug open to to ducumennt the qemu shortcoming for someone to work on, ...

mhdevsec, 2026-01-11:
I have taken a look at the issue.
This error is reproducible in qemu-sparc64 running on sparc64 architecture.
qemu-sparc64  /usr/bin/svn co http://svn.exactcode.de/t2/trunk t2-trunk

The problem seem to happen in /usr/lib64/libcrypto.so.3
https://github.com/openssl/openssl/blob/master/crypto/sparccpuid.S

_sparcv9_fmadd_probe:
	.word	0x81b00d80	!fxor	%f0,%f0,%f0
	.word	0x85b08d82	!fxor	%f2,%f2,%f2
	retl
	.word	0x81b80440	!fmaddd	%f0,%f0,%f2,%f0

[image]

At first I thought qemu didn't handle the delay slot correctly after the return,
or that the stack is totally in a mess.

But now I incline to believe that it really cannot handle the instruction.

fmaddd is performing a multiply + add instruction
r = a*b +c

Writing such code in C, will generate assembler code for both fmuld and faddd

 fmuld  %f8, %f12, %f8
 faddd  %f8, %f10, %f8

If I manually rewrite the .S file to conform to fmaddd instead
fmaddd  %f8, %f12, %f10, %f8

The following is produced when compiling with gcc
gcc version 15.2.0 (T2 SDE) 
COLLECT_GCC_OPTIONS='-v' '-o' 'fp_01' '-mcpu=ultrasparc3' '-mtune=niagara4'
 /usr/lib64/gcc/sparc64-t2-linux-gnu/15.2.0/../../../../sparc64-t2-linux-gnu/bin/as -v -s -Av9b -64 -no-undeclared-regs -relax -o /tmp/ccjVXlsu.o fp_01.s
GNU assembler version 2.45 (sparc64-t2-linux-gnu) using BFD version (GNU Binutils) 2.45
fp_01.s: Assembler messages:
fp_01.s:48: Error: Architecture mismatch on "fmaddd %f8,%f12,%f10,%f8".
fp_01.s:48: (Requires v9d|v9e|v9v|v9m|m8; requested architecture is v9b.)

v9b? what?
https://gcc.gnu.org/onlinedocs/gcc-13.1.0/gcc/SPARC-Options.html
The fmaddd instruction is a fairly "new" instruction it seems.

I am not sure what this fmadd_probe in openssl is doing.
If they are testing for the presence of the fmadd instruction or measuring the time of it.
but they have chosen a v9 instruction that is not present in all v9 versions.
That is my take on it, and qemu might not support the fmadd instruction

I have a bit of a trouble testing my hypothesis. Rewrite the fmadd to fmuld and faddd or nopping the instructions.
I tried to rebuild openssl but sshd fail to start, complaining my lib was newer than it was linked with.
I have dug myself into hole here. A small change turns out to be hard to test for me as I am new in building and testing T2.

Am I on to something regarding the instruction?

rxrbln, 2026-01-13:
on a closer look I overlooked in your initial post that gdb just stops on the sigill signal, you need to continue. I just did this and the segfault is later, after handling the 3 in my case illegal instructions so the actual problem is further down the code