rxrbln/t2sde #237: Build error on Ubuntu

ukrcherry, 2025-04-23:
Would there be any chance to do the cross-build on Ubuntu or doing that on a local T2SDE instance is the only option?

Ubuntu 24.04.02

`svn co https://svn.exactcode.de/t2/trunk t2-trunk
./t2 config --cfg system
./t2 download --cfg system --required`

success

`./t2 build-target --cfg system`

failure

`sudo ./t2 build-target --cfg system
[sudo] password for alex: 

Found 34 new/updated files in source tree:
  Found new/updated file: package/audio/jack2/jack2.desc
  Found new/updated file: package/audio/tenacity/tenacity.desc
  Found new/updated file: package/java/gradle/gradle.desc
  ** list truncated to save space on the terminal **

Re-running configuration:
  T2 25-svn configuration
  Aquiring config lock ...
  Lua Bash accel. (C) 2006-2021 by V. Ziegler, R. Rebe, ExactCODE GmbH
  New config written to: config/system/

Building tools in build/.../TOOLCHAIN/cross:
  Building cross/wrapper/cmd_wrapper
misc/tools-source/cmd_wrapper.c: In function 'main':
...
 Testing cross/lib/fl_wrapper.so
  Creating cross/lib/parse-config
  Creating cross/lib/pkg_*_{pre,post}.conf

Removing old error logs ...
  Removing var/adm/logs/0-gzip.err ...

18:01:30 6/5762 Building 0-archiver/gzip (1.14)
  Building in src.gzip.system.250423.180128.1611032, with 8 threads
mount: /tmp: mount point not mounted or bad option.
       dmesg(1) may have more information after failed mount system call.
  Writing output to $root/var/adm/logs/0-gzip.out
! checking whether the C compiler works... no
! configure: error: in '/home/alex/git/t2/t2src/src.gzip.system.250423.180128.1611032/gzip-1.14':
! configure: error: C compiler cannot create executables
! See 'config.log' for more details
! scripts/functions.in: line 974: pop_var_context: head of shell_variables not a function context
! scripts/Build-Pkg: line 670: pop_var_context: head of shell_variables not a function context
! Due to previous errors, no 0-gzip.log file!
! (Try enabling xtrace in the config to track an error inside the build system.)
  +00:00:14 Aborted building archiver/gzip
`

coolluke1992, 2025-04-23:
try sudo su

ukrcherry, 2025-04-23:
the same story

ˇsudo suˇ
ˇ./t2 config --cfg systemˇ
ˇ./t2 download --cfg system --requiredˇ
success

ˇ./t2 build-target --cfg systemˇ
failure

Building in src.gzip.system.250424.003515.1745730, with 8 threads
mount: /tmp: mount point not mounted or bad option.
       dmesg(1) may have more information after failed mount system call.
  Writing output to $root/var/adm/logs/0-gzip.out
! checking whether the C compiler works... no
! configure: error: in '/home/alex/git/t2/t2src/src.gzip.system.250424.003515.1745730/gzip-1.14':
! configure: error: C compiler cannot create executables
! See 'config.log' for more details
! scripts/functions.in: line 974: pop_var_context: head of shell_variables not a function context
! scripts/Build-Pkg: line 670: pop_var_context: head of shell_variables not a function context
! Due to previous errors, no 0-gzip.log file!
! (Try enabling xtrace in the config to track an error inside the build system.)
  +00:00:03 Aborted building archiver/gzip

coolluke1992, 2025-04-23:
have you tried the `curl -Ls https://t2linux.com/install | sh` in extract mode so then you can mount --bind proc and friends and a tmpfs then you can chroot . bin/bash

ukrcherry, 2025-04-24:
No, I have not. Normally, I avoid doing that kind of messing-up routine on the main work machine. But now I can see that the T2SDE cross compiling does not work as easy as i.e. pkgsrc/NetBSD croscompiling. It calls for having a dedicated machine where I have T2 installed or another test lab instance of Ubuntu.