rxrbln/t2sde #241: Correct way to rebuild glibc

NAHTAIV3L, 2025-06-25:
What is the correct way to rebuild glibc? Whenever I rebuild it I end up bricking my system. It seems to cause a ton of segfaults when installing the files then I can't use anything.

rxrbln, 2025-06-27:
thank you for your report, unfortunately I noticed this too. 20y ago we fixed source rebuild a couple of time, often it was glibc makefiles to use non atomic copies to install files. I'll try to check later today if we can make this work again. A workaround would be to use a temp DESTDIR and install the result ourselves. That would be a general cleaner staged building for other pkgs, too. but let me see what I can find.

lenticularis39, 2025-07-11:
There are also some (rare) packages that do not install if files are already present.

PrakarshPanwar, 2026-10-10:
Any update on this? Happened multiple times on my QEMU virtual machine. Always have to boot ISO and copy libc files to recover system.